ifm/libo3d3xx #75: Working with different IP addresses and OpenCV issue

Dalton0022, 2016-11-15:
I have been working with your libo3d3xx library for a few days trying to get it it interface with the 03d303 camera we have. I finally got it installed and have the example codes compiling and the command line utilities showing up. However, I can't seem to get connected to the camera as none of the examples or utils are working. I saw somewhere that the libraries and codes are assuming the default ip address for the camera and think/hope that may be the last problem I'm having since the ip address on our camera is set to 192.168.1.69:       
 _____________________________________________________________________________
   IP            At MAC Address      Count  Len   MAC Vendor                  
 -----------------------------------------------------------------------------
 192.168.1.69    00:02:01:40:2e:58    07    420   IFM Electronic gmbh  

I tried changing the ip address in a few places within the source code without any success so was wondering if you could quickly point me in the right direction to make this modification. Or maybe there is something else I should try? The viewer command line utility is also saying that openCV was built without openGL support. What would be the fastest way to solve this issue?

Thank you,

tpanzarella, 2016-11-15:
Depending on what you are trying to accomplish, using the non-default IP address can be utilized by the library in (at least) three different ways:

1. **Command-line utilities:**
All of the `o3d3xx-XXX` command-line utilities accept an `--ip` argument where you can specify your Camera's IP address explicitly. (ASIDE: All commands also accept `--help` listing all of their accepted arguments).

2. **Environment variable:**
At the library level, the environment variable `O3D3XX_IP` is respected. So, if you set that in your current environment, the library will assume to use the IP address specified in that envrionment variable.

3. **In code:**
The first argument to the `o3d3xx::Camera` ctor is the IP address:
```
o3d3xx::Camera::Ptr cam = std::make_shared<o3d3xx::Camera>("192.168.1.69");
```

The above should work for you.

> The viewer command line utility is also saying that openCV was built without openGL support. What would be the fastest way to solve this issue?

See if [this](https://github.com/lovepark/libo3d3xx/issues/56) helps.

Dalton0022, 2016-11-16:
Okay so I guess that's not the whole problem. I had tried your suggestion #3 and tried it again just to make sure and still had no success. I also tried the command line utility options which still reports "Timeout waiting for camera!" I installed IFM Vision Assistant on my windows computer and restored the camera to factory settings to see if that helped but still nothing (new ip address): 192.168.0.69. 

I'm now trying to remake the entire project from the top running all tests and this is the output I'm getting:

 [makelogfile.txt](https://github.com/lovepark/libo3d3xx/files/595181/makelogfile.txt)

Seems none of the checks are passing even with the default settings being used... any ideas?

Thanks,

tpanzarella, 2016-11-16:
Sounds like a more fundamental networking / connectivity issue. Please paste in the output of the following Linux commands:

```
$ ifconfig
```

```
$ netstat -r
```

```
$ ping 192.168.0.69
```

tpanzarella, 2016-11-16:
Also, please run:

```
$ o3d3xx-ls
```

Which, per your note above, should fail with the message:

```
Failed to list applications: XMLRPC call timed out
```

Then paste in the output of the logfile, you should see something like:

```
$ cat /tmp/libo3d3xx.INFO 
Log file created at: 2016/11/16 11:58:12
Running on machine: tuna
Log line format: [IWEF]mmdd hh:mm:ss.uuuuuu threadid file:line] msg
E1116 11:58:12.401237 12147 camera.hpp:797] http://192.168.0.69:80/api/rpc/v1/com.ifm.efector/ -> getAllParameters : Unable to transport XML to server and get XML response back.  libcurl failed to execute the HTTP POST transaction, explaining:  Connection timed out after 3001 milliseconds
```

This will show us explicitly which XML-RPC URL the library is trying to hit to reach your camera.

Dalton0022, 2016-11-16:
`$ ifconfig
eth0      Link encap:Ethernet  HWaddr b8:ae:ed:79:5e:fd  
          inet addr:192.168.100.101  Bcast:192.168.100.255  Mask:255.255.255.0
          inet6 addr: fe80::baae:edff:fe79:5efd/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:311 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3907 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:20008 (20.0 KB)  TX bytes:250300 (250.3 KB)
          Interrupt:20 Memory:aa100000-aa120000 

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:5250 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5250 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:637055 (637.0 KB)  TX bytes:637055 (637.0 KB)

wlan0     Link encap:Ethernet  HWaddr 94:65:9c:16:33:11  
          inet addr:10.33.28.64  Bcast:10.33.31.255  Mask:255.255.252.0
          inet6 addr: fe80::9665:9cff:fe16:3311/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:120850 errors:0 dropped:0 overruns:0 frame:0
          TX packets:13249 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:18700160 (18.7 MB)  TX bytes:1748284 (1.7 MB)

$ netstat -r
Kernel IP routing table
Destination     Gateway         Genmask         Flags   MSS Window  irtt Iface
default         10.33.31.254    0.0.0.0         UG        0 0          0 wlan0
10.33.28.0      *               255.255.252.0   U         0 0          0 wlan0
192.168.100.0   *               255.255.255.0   U         0 0          0 eth0
PING 192.168.0.69 (192.168.0.69) 56(84) bytes of data.

$ ping 192.168.0.69
--- 192.168.0.69 ping statistics ---
7 packets transmitted, 0 received, 100% packet loss, time 6040ms

$ o3d3xx-ls
Failed to list applications: XMLRPC call timed out

$ cat /tmp/libo3d3xx.INFO 
Log file created at: 2016/11/16 11:48:12
Running on machine: nuc-desktop
Log line format: [IWEF]mmdd hh:mm:ss.uuuuuu threadid file:line] msg
E1116 11:48:12.738246  9242 camera.hpp:797] http://192.168.0.69:80/api/rpc/v1/com.ifm.efector/ -> getParameter : Unable to transport XML to server and get XML response back.  libcurl failed to execute the HTTP POST transaction, explaining:  Connection timed out after 3001 milliseconds
E1116 11:48:12.739012  9242 frame_grabber.cpp:64] Could not get IP/Port of the camera: XMLRPC call timed out
W1116 11:48:12.739048  9242 frame_grabber.cpp:68] Assuming default PCIC port!
I1116 11:48:12.739069  9242 frame_grabber.cpp:72] Camera connection info: ip=192.168.0.69, port=50010
W1116 11:48:13.739276  9242 frame_grabber.cpp:193] Timeout waiting for image buffer from camera
W1116 11:48:13.739419  9244 frame_grabber.cpp:418] Exception: Thread interrupted
I1116 11:48:13.739444  9244 frame_grabber.cpp:421] Framegrabber thread done.`

tpanzarella, 2016-11-16:
Your network is set up incorrectly -- you have no route to the camera. If you cannot `ping` the camera, you will not be able to use `libo3d3xx` to interface to it. I recommend giving your hardwired ethernet port a static ip address on the same subnet as the camera, then plug the camera directly into the network port. Then, try to `ping` the camera. Once you can do that, then try using `libo3d3xx` to talk with the camera. If at that point, you are still having issues, please report back with what you are seeing.

Dalton0022, 2016-11-16:
Problem solved! I ended up setting the camera IP to match the computer's since the computer was already statically configured to work with two other cameras. Thank you, you help is much appreciated!

tpanzarella, 2016-11-16:
No worries. Best of luck.

closed 2016-11-16 by tpanzarella