ifm/libo3d3xx #76: Time to get the amplitude image data from the sensor

pavantankasala, 2016-11-17:
It is taking 50-100 ms to get the data from IFM o3d3xx sensor when operated in trigger mode. Is there a way to speed the process of acquiring the data in trigger mode.

graugans, 2016-11-17:
This depends heavily on your configuration and the network connection you are using. May I ask how you trigger the device and how you measure the delay?

pavantankasala, 2016-11-17:
we use a gigabit switch to communicate with the camera. we use a software trigger for triggering the camera. we measure the delay from the time of the trigger to the time we read the data in the callback

tpanzarella, 2016-11-17:
Could you also supply the output of `o3d3xx-dump`? Your integration times (and other imager parameters) will have significant influence on this metric.

pavantankasala, 2016-11-17:
thanks for the prompt response. Please find the dump below
status: 0
config: {
    "o3d3xx":
    {
        "libo3d3xx": "400",
        "Date": "Thu Nov 17 13:43:44 2016",
        "HWInfo":
        {
            "Mainboard": "#!02_M381_003_02165132_008022954",
            "Illumination": "#!02_I300_001_02100581_008001175",
            "MiraSerial": "3410-50b0-13e4-04c0",
            "MACAddress": "00:02:01:40:28:0D",
            "Frontend": "#!02_F341_002_15_00260_802259343",
            "Diagnose": "#!02_D322_C30_02101064_008023093",
            "Connector": "#!02_A300_001_02064651_008001175"
        },
        "SWVersion":
        {
            "IFM_Software": "1.4.1639",
            "Linux": "Linux version 3.11.0-00001-gda3d004 (eswadmin@dettlx79) (gcc version 4.7.2 (GCC) ) #1 SMP Wed Feb 17 15:55:11 CET 2016",
            "Diagnostic_Controller": "0.3.150",
            "ELDK": "GOLDENEYE_ELDK-2016-03-17_16-21-02-8baf2bee4a4eebca0d2ad7c7345d3a4f971032ec",
            "Calibration_Version": "0.9.0",
            "Calibration_Device": "00:02:01:40:28:0d",
            "Main_Application": "0.4.164",
            "Algorithm_Version": "1.2.14"
        },
        "Device":
        {
            "Name": "New sensor",
            "Description": "",
            "ActiveApplication": "1",
            "PcicEipEnabled": "false",
            "PcicTcpPort": "50010",
            "PcicProtocolVersion": "3",
            "IOLogicType": "1",
            "IODebouncing": "true",
            "IOExternApplicationSwitch": "0",
            "SessionTimeout": "30",
            "ServiceReportPassedBuffer": "15",
            "ServiceReportFailedBuffer": "15",
            "ExtrinsicCalibTransX": "0",
            "ExtrinsicCalibTransY": "0",
            "ExtrinsicCalibTransZ": "0",
            "ExtrinsicCalibRotX": "0",
            "ExtrinsicCalibRotY": "0",
            "ExtrinsicCalibRotZ": "0",
            "EvaluationFinishedMinHoldTime": "10",
            "SaveRestoreStatsOnApplSwitch": "true",
            "IPAddressConfig": "0",
            "PasswordActivated": "false",
            "OperatingMode": "0",
            "DeviceType": "",
            "ArticleNumber": "",
            "ArticleStatus": "",
            "UpTime": "6.91858491883055e-310",
            "ImageTimestampReference": "604008384",
            "TemperatureFront1": "6.918634422112782e-310",
            "TemperatureFront2": "6.918584918854265e-310",
            "TemperatureIMX6": "6.918584918866122e-310",
            "TemperatureIllu": "6.918600835395051e-310"
        },
        "Net":
        {
            "MACAddress": "",
            "NetworkSpeed": "0",
            "StaticIPv4Address": "10.10.30.75",
            "StaticIPv4SubNetMask": "255.255.255.0",
            "StaticIPv4Gateway": "10.10.30.0",
[... 85 lines cut ...]
                    "Resolution": "0",
                    "ClippingCuboid": "{\"XMin\": -3.402823e+38, \"XMax\": 3.402823e+38, \"YMin\": -3.402823e+38, \"YMax\": 3.402823e+38, \"ZMin\": -3.402823e+38, \"ZMax\": 3.402823e+38}",
                    "ReduceMotionArtifacts": "false",
                    "SpatialFilterType": "0",
                    "SymmetryThreshold": "0",
                    "TemporalFilterType": "0",
                    "ThreeFreqMax2FLineDistPercentage": "80",
                    "ThreeFreqMax3FLineDistPercentage": "80",
                    "TwoFreqMaxLineDistPercentage": "80",
                    "Type": "under5m_low",
                    "SpatialFilter":
                    {
                        "Type": "0",
                        "TypeStr_": "Off"
                    },
                    "TemporalFilter":
                    {
                        "Type": "0",
                        "TypeStr_": "Off"
                    }
                }
            }
        ]
    }
}

graugans, 2016-11-18:
Looks like you are using the mode ``"Type": "upto30m_moderate",`` In this mode the camera takes 4 images. 

``
for(freq = [f1,f2])
{
      for(exposure=[short,long])
      {
         exposure_image();
         pre_process();
      }
}
calculate_3d_point_cloud()
```
This is not the mode with the lowest latency. It heavily depends on your scene, maybe you can switch to a mode like ``under5m_low`` or if you need a higher scan rate use the free-running mode. For details about the available modes I recommend to read [the manual](http://www.ifm.com/mounting/706395UK.pdf) 7.3.2 Set maximum background distance.

cfreundl, 2016-11-19:
I did some rough calculations for the exposure modes listed in your camera configuration, and I dare to guess that the 50-100 ms you measured did not use application 1 but application 2, right?

To elaborate a little more on what @graugans already said, each "image" taken by the camera consists itself of 4 phases, and the exposure times you configure are those for a single phase. Together with some additional required phases for the whole exposure sequence I can compute the rough overall acquisition time of application 1 to
````
2 [frequencies] * (
    4 [phases] * 10 ms [long exposure] +
    4 [phases] * 0.25 ms [short exposure]
) +
1 ms [warmup phase] +
19 [total phases] * 2ms [readout time]
= 121 ms
````
and the acquisition time of application 2 to
```
4 [phases] * 1 ms [exposure] +
1 ms [warmup phase] +
6 [total phases] * 2 ms [readout time]
= 17 ms
```
In addition to the exposure itself, you also have to take the time to transfer the data into the CPU into account for the overall acquisition time (the "readout times").

The computation of the distance images is the next time consuming operation on the camera. For application 1 I would refrain from giving an estimate, because I do not know the pure computation time by heart (and also the computation overlaps with the acquisition), but I can catch up later if desired. For application 2 I would give a rough estimate of 40 ms computation time, and in that exposure mode, computation strictly takes place after the acquisition, as the is only a single "image". That means, in that mode alone the camera's acquisition and computation contribute at least 57 ms to the latency you are seeing. Add to that some trigger command processing and result transfer preparation on the camera plus two times the network latency, and you have your measurements explained.

closed 2020-07-27 by theseankelly