ifm/libo3d3xx #77: Build errors

sroth44489, 2017-01-10:
I have a new install of Ubuntu Trusty (14.04) with ros indigo, and am running into errors on the unit tests when building the libo3d3xx library. I have a brand new O3D303 running firmware 1.6.2038.  

When running make, most unit tests pass, but the following 5 fail:

[  FAILED  ] 5 tests, listed below:
[  FAILED  ] AppImagerTest.GetImagerParameters
[  FAILED  ] AppImagerTest.ImagerConfig
[  FAILED  ] Camera_Tests.GetDeviceConfig
[  FAILED  ] Camera_Tests.SetDeviceConfig
[  FAILED  ] Camera_Tests.TemporaryParameters

Here are the individual error messages:

[ RUN      ] AppImagerTest.GetImagerParameters
/home/sroth/libo3d3xx/modules/camera/test/o3d3xx-app-imager-tests.cpp:270: Failure
Expected: params.at("UseSimpleBinning") doesn't throw an exception.
  Actual: it throws.
[  FAILED  ] AppImagerTest.GetImagerParameters (2177 ms)

[ RUN      ] AppImagerTest.ImagerConfig
/home/sroth/libo3d3xx/modules/camera/test/o3d3xx-app-imager-tests.cpp:420: Failure
Expected: cam_->SetImagerConfig(im.get()) doesn't throw an exception.
  Actual: it throws.
[  FAILED  ] AppImagerTest.ImagerConfig (2266 ms)

[ RUN      ] Camera_Tests.GetDeviceConfig
/home/sroth/libo3d3xx/modules/camera/test/o3d3xx-camera-tests.cpp:182: Failure
Value of: 34
Expected: params.size()
Which is: 33
unknown file: Failure
C++ exception with description "_Map_base::at" thrown in the test body.
[  FAILED  ] Camera_Tests.GetDeviceConfig (880 ms)

[ RUN      ] Camera_Tests.SetDeviceConfig
/home/sroth/libo3d3xx/modules/camera/test/o3d3xx-camera-tests.cpp:269: Failure
Expected: cam->SetDeviceConfig(dev.get()) doesn't throw an exception.
  Actual: it throws.
/home/sroth/libo3d3xx/modules/camera/test/o3d3xx-camera-tests.cpp:276: Failure
Value of: dev->EthernetFieldBusEndianness()
  Actual: 0
Expected: tmp_eth_field_bus_endianness
Which is: 1
[  FAILED  ] Camera_Tests.SetDeviceConfig (2300 ms)

[ RUN      ] Camera_Tests.TemporaryParameters
unknown file: Failure
C++ exception with description "XMLRPC method not found" thrown in the test body.
[  FAILED  ] Camera_Tests.TemporaryParameters (44 ms)

Is this a problem with the newest firmware?

Thanks much for this library
Stephan

sroth44489, 2017-01-10:
Hi Tom,
  Everything's going great. Hope everything's well with you. Thanks so much for your quick help. The version 0.4.8 builds now.

Thanks,
Stephan

closed 2017-01-10 by sroth44489

mikeNike9, 2017-01-31:
I had same error and 0.4.8 fixed this issue but FrameGrabberTest is failing everytime.

/home/hayden/libo3d3xx-0.4.8/modules/framegrabber/test/o3d3xx-frame_grabber-tests.cpp:52: Failure
Value of: fg->WaitForFrame(buff.get(), 1000)
  Actual: false
Expected: true
/home/hayden/libo3d3xx-0.4.8/modules/framegrabber/test/o3d3xx-frame_grabber-tests.cpp:52: Failure
Value of: fg->WaitForFrame(buff.get(), 1000)
  Actual: false
Expected: true
/home/hayden/libo3d3xx-0.4.8/modules/framegrabber/test/o3d3xx-frame_grabber-tests.cpp:52: Failure
Value of: fg->WaitForFrame(buff.get(), 1000)
  Actual: false
Expected: true
/home/hayden/libo3d3xx-0.4.8/modules/framegrabber/test/o3d3xx-frame_grabber-tests.cpp:52: Failure
Value of: fg->WaitForFrame(buff.get(), 1000)
  Actual: false
Expected: true
/home/hayden/libo3d3xx-0.4.8/modules/framegrabber/test/o3d3xx-frame_grabber-tests.cpp:52: Failure
Value of: fg->WaitForFrame(buff.get(), 1000)
  Actual: false
Expected: true
/home/hayden/libo3d3xx-0.4.8/modules/framegrabber/test/o3d3xx-frame_grabber-tests.cpp:52: Failure
Value of: fg->WaitForFrame(buff.get(), 1000)
  Actual: false
Expected: true
/home/hayden/libo3d3xx-0.4.8/modules/framegrabber/test/o3d3xx-frame_grabber-tests.cpp:52: Failure
Value of: fg->WaitForFrame(buff.get(), 1000)
  Actual: false
Expected: true
/home/hayden/libo3d3xx-0.4.8/modules/framegrabber/test/o3d3xx-frame_grabber-tests.cpp:52: Failure
Value of: fg->WaitForFrame(buff.get(), 1000)
  Actual: false
Expected: true
/home/hayden/libo3d3xx-0.4.8/modules/framegrabber/test/o3d3xx-frame_grabber-tests.cpp:52: Failure
Value of: fg->WaitForFrame(buff.get(), 1000)
  Actual: false
Expected: true
/home/hayden/libo3d3xx-0.4.8/modules/framegrabber/test/o3d3xx-frame_grabber-tests.cpp:52: Failure
Value of: fg->WaitForFrame(buff.get(), 1000)
  Actual: false
Expected: true
[  FAILED  ] FrameGrabberTest.WaitForFrame (13173 ms)
[ RUN      ] FrameGrabberTest.WaitForFrameFast
/home/hayden/libo3d3xx-0.4.8/modules/framegrabber/test/o3d3xx-frame_grabber-tests.cpp:69: Failure
Value of: fg->WaitForFrame(buff.get(), 1000, false, false)
  Actual: false
Expected: true
/home/hayden/libo3d3xx-0.4.8/modules/framegrabber/test/o3d3xx-frame_grabber-tests.cpp:69: Failure
Value of: fg->WaitForFrame(buff.get(), 1000, false, false)
  Actual: false
Expected: true
/home/hayden/libo3d3xx-0.4.8/modules/framegrabber/test/o3d3xx-frame_grabber-tests.cpp:69: Failure
Value of: fg->WaitForFrame(buff.get(), 1000, false, false)
  Actual: false
Expected: true
/home/hayden/libo3d3xx-0.4.8/modules/framegrabber/test/o3d3xx-frame_grabber-tests.cpp:69: Failure
Value of: fg->WaitForFrame(buff.get(), 1000, false, false)
  Actual: false
Expected: true
/home/hayden/libo3d3xx-0.4.8/modules/framegrabber/test/o3d3xx-frame_grabber-tests.cpp:69: Failure
Value of: fg->WaitForFrame(buff.get(), 1000, false, false)
  Actual: false
Expected: true
/home/hayden/libo3d3xx-0.4.8/modules/framegrabber/test/o3d3xx-frame_grabber-tests.cpp:69: Failure
Value of: fg->WaitForFrame(buff.get(), 1000, false, false)
  Actual: false
Expected: true
/home/hayden/libo3d3xx-0.4.8/modules/framegrabber/test/o3d3xx-frame_grabber-tests.cpp:69: Failure
Value of: fg->WaitForFrame(buff.get(), 1000, false, false)
[... 84 lines cut ...]
[==========] 9 tests from 1 test case ran. (72873 ms total)
[  PASSED  ] 3 tests.
[  FAILED  ] 6 tests, listed below:
[  FAILED  ] FrameGrabberTest.WaitForFrame
[  FAILED  ] FrameGrabberTest.WaitForFrameFast
[  FAILED  ] FrameGrabberTest.CustomSchemas
[  FAILED  ] FrameGrabberTest.ByteBufferCopyCtor
[  FAILED  ] FrameGrabberTest.ByteBufferCopyAssignmentOperator
[  FAILED  ] FrameGrabberTest.FrameGrabberRecycling

 6 FAILED TESTS
test/CMakeFiles/check.dir/build.make:57: recipe for target 'test/CMakeFiles/check' failed
make[6]: *** [test/CMakeFiles/check] Error 1
CMakeFiles/Makefile2:369: recipe for target 'test/CMakeFiles/check.dir/all' failed
make[5]: *** [test/CMakeFiles/check.dir/all] Error 2
CMakeFiles/Makefile2:376: recipe for target 'test/CMakeFiles/check.dir/rule' failed
make[4]: *** [test/CMakeFiles/check.dir/rule] Error 2
Makefile:262: recipe for target 'check' failed
make[3]: *** [check] Error 2
CMakeFiles/O3D3XX_FRAMEGRABBER.dir/build.make:120: recipe for target 'O3D3XX_FRAMEGRABBER-prefix/src/O3D3XX_FRAMEGRABBER-stamp/O3D3XX_FRAMEGRABBER-test' failed
make[2]: *** [O3D3XX_FRAMEGRABBER-prefix/src/O3D3XX_FRAMEGRABBER-stamp/O3D3XX_FRAMEGRABBER-test] Error 2
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/O3D3XX_FRAMEGRABBER.dir/all' failed
make[1]: *** [CMakeFiles/O3D3XX_FRAMEGRABBER.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

tpanzarella, 2017-02-01:
All of these errors indicate a timeout between your machine and the PCIC server on the camera. Typical causes would be:

1. You don't have connectivity to the camera. Can you ping your camera from your machine?
2. Errors in parsing the return data from the camera are causing the frame grabber to not packetize the data properly before signalling waiting clients. The only thing I can think of that can cause this is a version of the library that is incompatible with the on-camera firmware version.
3. The FrameGrabber unit test assumes the camera is in "Free Running" mode, i.e., you do not need to software trigger an image acquisition. If the active application on the camera, at the time of running this test, is configured for software triggering, this timeout would be expected.
4. You have no application on the camera marked as active.

The above said, could you paste in the output of:

```
$ o3d3xx-dump
```

and

```
$ ifconfig
```

It would also be good to get a copy of the log file just after the unit test that is failing is done. The default location is: `/tmp/libo3d3xx.INFO`.

This should help us figure out what is going on.