ifm/libo3d3xx #78: pcl::PointCloud<o3d3xx::PointT>::Ptr cloud = img->Cloud();

mikeNike9, 2017-02-03:


tpanzarella, 2017-02-03:
Without any detail to this issue, there is no assistance we can provide. Closing as N/A.

mikeNike9, 2017-02-03:
i just edited my comment
sorry\

tpanzarella, 2017-02-03:
The coordinate frame of the point cloud is x-forward, y-left, z-up. ex-cartesian does that transform [here](https://github.com/lovepark/libo3d3xx/blob/master/modules/examples/ex-cartesian.cpp#L108)

mikeNike9, 2017-02-03:
understandable.  Thanks for clearing that up.  but even when I figured that out it still does not return the same result.  center pixel is not the same.

tpanzarella, 2017-02-03:
More info, code, imagery would help.

One note: The point cloud is in meters. When computing the cartesian data from radial distance image, unit vectors, and extrinsic calibration that will be in mm. Watch your conversions between int16_t and float. Maybe this helps.

tpanzarella, 2017-02-03:
> In windows I can have 5 hnd open and trigger the 5 cameras in a row grab all the data and then process it and get a final measurement in about 2-3 seconds.

If I understand you correctly here, to achieve a similar thing in libo3d3xx, you will simply want to instantiate new camera, framegrabber, and imagebuffer instances for each physical camera (discriminating each by IP address) and you can do the same thing. Creating a `std::future` to manage each connection would allow you to concurrently acquire and process the data.

mikeNike9, 2017-02-04:
Everything is working well now.
There were minor issues on how I was pointing to the cloud_

Now I can run my filters perfectly with two cameras.
I did what you said in the above step with invidual instances of the camera, framegrabber, and image buffer 
but I seem to only be able to run it with two cameras. As soon as I add a 3rd instance of each and try to run it the camera times out on the third  instance of img3->cloud.
I am having it wait 6000 ms for it to get the image but it still will time out.
The std::future is something I will be reaseraching this weekend. never have had to use this class.  
I am assuming that once I implement the std::future to handle all the instances then I will be able to get all of the data out of the camera..
Let me know what you think.
THank you

tpanzarella, 2017-02-04:
> Now I can run my filters perfectly with two cameras.
I did what you said in the above step with invidual instances of the camera, framegrabber, and image buffer
but I seem to only be able to run it with two cameras. As soon as I add a 3rd instance of each and try to run it the camera times out on the third instance of img3->cloud.
I am having it wait 6000 ms for it to get the image but it still will time out.

Show me some code, and I can likely help you. However, my intuition tells me, you are likely running this all in a single thread. And, IIRC, you are working in a software triggered environment instead of a free-running/streaming environment. With those assumptions in place, I also assume that you are likely calling `fg->SWTrigger()` on each one of your framegrabber instances then calling `WaitForFrame(...)` on each one in turn -- or some similar pattern. This could lead to the timeout you are seeing. Realize, under the hood, each framegrabber is running a separate worker thread from your main thread puling in and parsing the pixel data. So, when that initial series of `SWTrigger()` trigger calls are made, they are running concurrently. So, by the time you call `WaitForFrame` on the third instance, the triggered frame has already been parsed and (from one perspective) is too old to return from the call to `WaitForFrame` -- I suppose we could add a new call like `GetLatestFrame` but that does not exist in the library today. The way to fix your issue in a single-threaded mode like you are running in, would be to call `SWTrigger()` immediately followed by `WaitForFrame(...)` for each instance as a pair of calls.

> The std::future is something I will be reaseraching this weekend. never have had to use this class.
I am assuming that once I implement the std::future to handle all the instances then I will be able to get all of the data out of the camera..
Let me know what you think.

Sure. Whether you use `std::future` or explicitly spin up a thread pool, or employ some other concurrency idiom (I don't know the constraints of your application, so can't really give you a recommendation), you should be able to trigger and acquire pixel data from each camera concurrently. There is no theoretical limit in the library, however, I wouldn't spin up any more than the return from `std::thread::hardware_concurrency()`. You should also exercise some sensitivity to your network topology.

closed 2017-08-25 by tpanzarella