ifm/libo3d3xx #79: does not contain a CMakeList.txt file.

KtGunn, 2017-02-14:
I need help, please.
I get the error message in the title from the 'make' step of the build.
Looks like the build is expecting to find a cmake file in /usr/src/gtest, which I manually created,
but there is none there. I have not tried to build with tests disabled.
Thanks,
-Kris

tpanzarella, 2017-02-14:
If you are on Ubuntu you could `sudo apt-get install libgtest-dev` and that should solve your problem. 

Otherwise, you could edit the individual `modules/XXX/test/CMakeLists.txt` files in each of the `libo3d3xx` module subdirectories and edit the line `set(GTEST_CMAKE_DIR "/usr/src/gtest")` to point to your install of `gtest`. [GTest](https://github.com/google/googletest) is the unit testing framework employed by `libo3d3xx`.

KtGunn, 2017-02-14:
Thanks for your very quick response, 't'.
I took the sudo apt-get route. The build instructions in the github README.md are, well, vague.
And a cmake file did show up in /usr/src/gtest indeed, but I've run into another problem compiling:

.....
[ 75%] Building CXX object test/gtest_bin/CMakeFiles/gtest.dir/src/gtest-all.cc.o
In file included from /usr/local/include/gtest/internal/gtest-internal.h:64:0,
                 from /usr/local/include/gtest/gtest.h:58,
                 from /usr/src/gtest/src/gtest-all.cc:39:
/usr/local/include/gtest/internal/gtest-string.h: In constructor ‘testing::internal::GTestFlagSaver::GTestFlagSaver()’:
/usr/local/include/gtest/internal/gtest-string.h:157:3: error: ‘testing::internal::String::String()’ is private
   String();  // Not meant to be instantiated.
......
< more bad news>

Any thought?
Thanks.
-Kris

tpanzarella, 2017-02-14:
Looks like you have multiple versions of `gtest` installed and you are mixing them in the same build. See compiler error that is referencing both `/usr/src/gtest` and `/usr/local/include/gtest`. The Ubuntu deb does not install into `/usr/local/` see:

```
$ dpkg -L libgtest-dev
/.
/usr
/usr/share
/usr/share/doc
/usr/share/doc/libgtest-dev
/usr/share/doc/libgtest-dev/README.Debian
/usr/share/doc/libgtest-dev/examples
/usr/share/doc/libgtest-dev/examples/make
/usr/share/doc/libgtest-dev/examples/make/Makefile
/usr/share/doc/libgtest-dev/examples/samples
/usr/share/doc/libgtest-dev/examples/samples/sample4_unittest.cc
/usr/share/doc/libgtest-dev/examples/samples/sample2_unittest.cc
/usr/share/doc/libgtest-dev/examples/samples/sample4.h
/usr/share/doc/libgtest-dev/examples/samples/sample6_unittest.cc.gz
/usr/share/doc/libgtest-dev/examples/samples/sample8_unittest.cc.gz
/usr/share/doc/libgtest-dev/examples/samples/sample1.cc
/usr/share/doc/libgtest-dev/examples/samples/sample4.cc
/usr/share/doc/libgtest-dev/examples/samples/sample7_unittest.cc.gz
/usr/share/doc/libgtest-dev/examples/samples/prime_tables.h
/usr/share/doc/libgtest-dev/examples/samples/sample3_unittest.cc.gz
/usr/share/doc/libgtest-dev/examples/samples/sample2.h
/usr/share/doc/libgtest-dev/examples/samples/sample2.cc
/usr/share/doc/libgtest-dev/examples/samples/sample10_unittest.cc.gz
/usr/share/doc/libgtest-dev/examples/samples/sample5_unittest.cc.gz
/usr/share/doc/libgtest-dev/examples/samples/sample9_unittest.cc.gz
/usr/share/doc/libgtest-dev/examples/samples/sample1.h
/usr/share/doc/libgtest-dev/examples/samples/sample1_unittest.cc.gz
/usr/share/doc/libgtest-dev/examples/samples/sample3-inl.h.gz
/usr/share/doc/libgtest-dev/copyright
/usr/share/doc/libgtest-dev/README.gz
/usr/share/doc/libgtest-dev/changelog.Debian.gz
/usr/src
/usr/src/gtest
/usr/src/gtest/cmake
/usr/src/gtest/cmake/internal_utils.cmake
/usr/src/gtest/CMakeLists.txt
/usr/src/gtest/src
/usr/src/gtest/src/gtest-port.cc
/usr/src/gtest/src/gtest.cc
/usr/src/gtest/src/gtest-typed-test.cc
/usr/src/gtest/src/gtest-filepath.cc
/usr/src/gtest/src/gtest-internal-inl.h
/usr/src/gtest/src/gtest-all.cc
/usr/src/gtest/src/gtest-test-part.cc
/usr/src/gtest/src/gtest-printers.cc
/usr/src/gtest/src/gtest_main.cc
/usr/src/gtest/src/gtest-death-test.cc
/usr/include
/usr/include/gtest
/usr/include/gtest/gtest-param-test.h
/usr/include/gtest/gtest.h
/usr/include/gtest/gtest_prod.h
/usr/include/gtest/gtest-param-test.h.pump
/usr/include/gtest/gtest-typed-test.h
/usr/include/gtest/gtest-death-test.h
/usr/include/gtest/gtest-message.h
/usr/include/gtest/gtest-test-part.h
/usr/include/gtest/internal
/usr/include/gtest/internal/gtest-param-util-generated.h
/usr/include/gtest/internal/gtest-param-util.h
/usr/include/gtest/internal/gtest-type-util.h
/usr/include/gtest/internal/gtest-port.h
/usr/include/gtest/internal/gtest-string.h
/usr/include/gtest/internal/gtest-filepath.h
/usr/include/gtest/internal/gtest-tuple.h.pump
/usr/include/gtest/internal/gtest-linked_ptr.h
/usr/include/gtest/internal/gtest-param-util-generated.h.pump
/usr/include/gtest/internal/gtest-tuple.h
/usr/include/gtest/internal/gtest-type-util.h.pump
/usr/include/gtest/internal/gtest-internal.h
/usr/include/gtest/internal/gtest-death-test-internal.h
/usr/include/gtest/gtest-spi.h
/usr/include/gtest/gtest_pred_impl.h
/usr/include/gtest/gtest-printers.h
```

FWIW, I'm on 16.04.1 LTS.

KtGunn, 2017-02-14:
't', you're right. 

Because this was such an unexpected compile error, I suspected a mismatch (I had tried 3 different ways to install gtest!). I ran "sudo apt-get purge libdev-gtest," then re-installed. But there were still files found on the path in /usr/local as you noted. Once I deleted those, compilation succeeded. 
Thanks again.

I did have 6 failed test, by the way, 3 in each AppImagerTest and CameraTest. The failed ones were all the first tests in a series, and appear to be info and configuration tests. (Maybe a firmware upgrade is in order.)

Thanks again.

tpanzarella, 2017-02-14:
> I did have 6 failed test, by the way, 3 in each AppImagerTest and CameraTest. The failed ones were all the first tests in a series, and appear to be info and configuration tests. (Maybe a firmware upgrade is in order.)

Yes. Please check the software compatibility matrix in the [README](https://github.com/lovepark/libo3d3xx/blob/master/README.md) for supported library / firmware versions. In the near future, this coupling will be less rigid, but for now, you need to make sure you are running a supported pair.

KtGunn, 2017-02-14:
Will do. Thanks.

closed 2017-02-14 by KtGunn