ifm/libo3d3xx #81: Some problems trying to make the library.

RamonGomez, 2017-03-08:
Hello, 

Im getting those errors while trying to make:

root@ramon-Vostro-5568:/dev/libo3d3xx/modules/camera/build# make check
[  8%] Built target gtest
[ 16%] Built target gtest_main
[ 62%] Built target o3d3xx_camera
[ 66%] Built target o3d3xx_camera_shared
[100%] Built target o3d3xx-camera-tests
[==========] Running 44 tests from 6 test cases.
[----------] Global test environment set-up.
[----------] 4 tests from TemporalFilterTest
[ RUN      ] TemporalFilterTest.TemporalFilterConfig_General
[       OK ] TemporalFilterTest.TemporalFilterConfig_General (2048 ms)
[ RUN      ] TemporalFilterTest.GetTemporalFilterParameters
[       OK ] TemporalFilterTest.GetTemporalFilterParameters (1929 ms)
[ RUN      ] TemporalFilterTest.GetTemporalFilterParameterLimits
[       OK ] TemporalFilterTest.GetTemporalFilterParameterLimits (1792 ms)
[ RUN      ] TemporalFilterTest.TemporalFilterConfig_JSON
[       OK ] TemporalFilterTest.TemporalFilterConfig_JSON (1981 ms)
[----------] 4 tests from TemporalFilterTest (7752 ms total)

[----------] 4 tests from SpatialFilterTest
[ RUN      ] SpatialFilterTest.SpatialFilterConfig_General
[       OK ] SpatialFilterTest.SpatialFilterConfig_General (896 ms)
[ RUN      ] SpatialFilterTest.GetSpatialFilterParameters
[       OK ] SpatialFilterTest.GetSpatialFilterParameters (2278 ms)
[ RUN      ] SpatialFilterTest.GetSpatialFilterParameterLimits
[       OK ] SpatialFilterTest.GetSpatialFilterParameterLimits (1937 ms)
[ RUN      ] SpatialFilterTest.SpatialFilterConfig_JSON
[       OK ] SpatialFilterTest.SpatialFilterConfig_JSON (1903 ms)
[----------] 4 tests from SpatialFilterTest (7014 ms total)

[----------] 1 test from ImportExport_Tests
[ RUN      ] ImportExport_Tests.ImportExportApp
[       OK ] ImportExport_Tests.ImportExportApp (1377 ms)
[----------] 1 test from ImportExport_Tests (1377 ms total)

[----------] 18 tests from Camera_Tests
[ RUN      ] Camera_Tests.Ctor
[       OK ] Camera_Tests.Ctor (3 ms)
[ RUN      ] Camera_Tests.GetXMLRPCURLPrefix
[       OK ] Camera_Tests.GetXMLRPCURLPrefix (0 ms)
[ RUN      ] Camera_Tests.GetAllParameters
[       OK ] Camera_Tests.GetAllParameters (247 ms)
[ RUN      ] Camera_Tests.GetParameter
[       OK ] Camera_Tests.GetParameter (1134 ms)
[ RUN      ] Camera_Tests.GetSWVersion
[       OK ] Camera_Tests.GetSWVersion (19 ms)
[ RUN      ] Camera_Tests.GetApplicationList
[       OK ] Camera_Tests.GetApplicationList (12 ms)
[ RUN      ] Camera_Tests.RequestSession
[       OK ] Camera_Tests.RequestSession (34 ms)
[ RUN      ] Camera_Tests.CancelSession
[       OK ] Camera_Tests.CancelSession (43 ms)
[ RUN      ] Camera_Tests.Heartbeat
[       OK ] Camera_Tests.Heartbeat (73 ms)
[ RUN      ] Camera_Tests.SetOperatingMode
[       OK ] Camera_Tests.SetOperatingMode (506 ms)
[ RUN      ] Camera_Tests.GetDeviceConfig
/dev/libo3d3xx/modules/camera/test/o3d3xx-camera-tests.cpp:182: Failure
Value of: 34
Expected: params.size()
Which is: 32
unknown file: Failure
C++ exception with description "_Map_base::at" thrown in the test body.
[  FAILED  ] Camera_Tests.GetDeviceConfig (688 ms)
[ RUN      ] Camera_Tests.ActivateDisablePassword
[       OK ] Camera_Tests.ActivateDisablePassword (459 ms)
[ RUN      ] Camera_Tests.SetDeviceConfig
/dev/libo3d3xx/modules/camera/test/o3d3xx-camera-tests.cpp:269: Failure
Expected: cam->SetDeviceConfig(dev.get()) doesn't throw an exception.
  Actual: it throws.
/dev/libo3d3xx/modules/camera/test/o3d3xx-camera-tests.cpp:275: Failure
Value of: dev->EthernetFieldBus()
  Actual: 0
Expected: tmp_eth_field_bus
Which is: 2
/dev/libo3d3xx/modules/camera/test/o3d3xx-camera-tests.cpp:276: Failure
Value of: dev->EthernetFieldBusEndianness()
  Actual: 0
Expected: tmp_eth_field_bus_endianness
Which is: 1
[  FAILED  ] Camera_Tests.SetDeviceConfig (2770 ms)
[ RUN      ] Camera_Tests.DeviceConfig_JSON
[       OK ] Camera_Tests.DeviceConfig_JSON (271 ms)
[ RUN      ] Camera_Tests.GetNetParameters
[       OK ] Camera_Tests.GetNetParameters (403 ms)
[ RUN      ] Camera_Tests.NetConfig
[       OK ] Camera_Tests.NetConfig (462 ms)
[ RUN      ] Camera_Tests.NetConfig_JSON
[       OK ] Camera_Tests.NetConfig_JSON (452 ms)
[ RUN      ] Camera_Tests.TemporaryParameters
unknown file: Failure
C++ exception with description "XMLRPC method not found" thrown in the test body.
[  FAILED  ] Camera_Tests.TemporaryParameters (44 ms)
[----------] 18 tests from Camera_Tests (7622 ms total)

[----------] 2 tests from General_Tests
[ RUN      ] General_Tests.Exceptions
[       OK ] General_Tests.Exceptions (0 ms)
[ RUN      ] General_Tests.Version
[       OK ] General_Tests.Version (0 ms)
[----------] 2 tests from General_Tests (0 ms total)

[----------] 15 tests from AppImagerTest
[ RUN      ] AppImagerTest.CopyDeleteApplication
[       OK ] AppImagerTest.CopyDeleteApplication (1259 ms)
[ RUN      ] AppImagerTest.CreateApplication
[       OK ] AppImagerTest.CreateApplication (866 ms)
[ RUN      ] AppImagerTest.ChangeAppNameAndDescription
[       OK ] AppImagerTest.ChangeAppNameAndDescription (1018 ms)
[ RUN      ] AppImagerTest.EditApplication
[       OK ] AppImagerTest.EditApplication (1131 ms)
[ RUN      ] AppImagerTest.GetAppParameters
/dev/libo3d3xx/modules/camera/test/o3d3xx-app-imager-tests.cpp:131: Failure
Value of: 9
Expected: params.size()
Which is: 8
[  FAILED  ] AppImagerTest.GetAppParameters (1540 ms)
[ RUN      ] AppImagerTest.AppConfig
[       OK ] AppImagerTest.AppConfig (1876 ms)
[ RUN      ] AppImagerTest.AppConfig_JSON
[       OK ] AppImagerTest.AppConfig_JSON (1691 ms)
[ RUN      ] AppImagerTest.GetAvailableImagerTypes
[       OK ] AppImagerTest.GetAvailableImagerTypes (1488 ms)
[ RUN      ] AppImagerTest.ChangeImagerType
[       OK ] AppImagerTest.ChangeImagerType (2305 ms)
[ RUN      ] AppImagerTest.GetImagerParameters
/dev/libo3d3xx/modules/camera/test/o3d3xx-app-imager-tests.cpp:270: Failure
Expected: params.at("UseSimpleBinning") doesn't throw an exception.
  Actual: it throws.
[  FAILED  ] AppImagerTest.GetImagerParameters (1737 ms)
[ RUN      ] AppImagerTest.GetImagerParameterLimits
[       OK ] AppImagerTest.GetImagerParameterLimits (2377 ms)
[ RUN      ] AppImagerTest.ImagerConfig
/dev/libo3d3xx/modules/camera/test/o3d3xx-app-imager-tests.cpp:420: Failure
Expected: cam_->SetImagerConfig(im.get()) doesn't throw an exception.
  Actual: it throws.
[  FAILED  ] AppImagerTest.ImagerConfig (2102 ms)
[ RUN      ] AppImagerTest.ImagerConfigValueOutOfRange
[       OK ] AppImagerTest.ImagerConfigValueOutOfRange (2439 ms)
[ RUN      ] AppImagerTest.ImagerConfig_JSON
[       OK ] AppImagerTest.ImagerConfig_JSON (1568 ms)
[ RUN      ] AppImagerTest.Exposure
[       OK ] AppImagerTest.Exposure (2804 ms)
[----------] 15 tests from AppImagerTest (26204 ms total)

[----------] Global test environment tear-down
[==========] 44 tests from 6 test cases ran. (49969 ms total)
[  PASSED  ] 38 tests.
[  FAILED  ] 6 tests, listed below:
[  FAILED  ] Camera_Tests.GetDeviceConfig
[  FAILED  ] Camera_Tests.SetDeviceConfig
[  FAILED  ] Camera_Tests.TemporaryParameters
[  FAILED  ] AppImagerTest.GetAppParameters
[  FAILED  ] AppImagerTest.GetImagerParameters
[  FAILED  ] AppImagerTest.ImagerConfig

 6 FAILED TESTS
test/CMakeFiles/check.dir/build.make:57: fallo en las instrucciones para el objetivo 'test/CMakeFiles/check'
make[3]: *** [test/CMakeFiles/check] Error 1
CMakeFiles/Makefile2:681: fallo en las instrucciones para el objetivo 'test/CMakeFiles/check.dir/all'
make[2]: *** [test/CMakeFiles/check.dir/all] Error 2
CMakeFiles/Makefile2:688: fallo en las instrucciones para el objetivo 'test/CMakeFiles/check.dir/rule'
make[1]: *** [test/CMakeFiles/check.dir/rule] Error 2
Makefile:366: fallo en las instrucciones para el objetivo 'check'
make: *** [check] Error 2

I'm new using Ubuntu (16.04 btw) and dont know what can be, maybe a lost library?

Greetings and lot of thanks.
Juan Ramón Gómez Martín.

tpanzarella, 2017-03-08:
This is indicative of a mismatch in firmware version to library version. Please see the top level README for a s/w compatibility table. Be sure to use a supported pair. 

Tom Panzarella
Founder & CEO
Love Park Robotics, LLC
2025 Washington Ave, Suite 217
Philadelphia, PA 19146 USA

mobile: +1 215 341 1754
tom@loveparkrobotics.com
http://loveparkrobotics.com

RamonGomez, 2017-03-08:
It worked but i still getting the error
"C++ exception with description "XMLRPC method not found" thrown in the test body." 
I just updated to the last version of the IFM firmware. Should i downgrade to a stable version?

Thanks again.
Juan Ramón Gómez Martín.

tpanzarella, 2017-03-08:
A table mapping compatible pairs of the library to ifm firmware version is [here](https://github.com/lovepark/libo3d3xx)

Tagged releases of the library are [here](https://github.com/lovepark/libo3d3xx/releases)

Older firmware versions are available [here](https://github.com/lovepark/o3d3xx-firmware) or also likely through ifm (I am not sure).

Please use a supported pair.

There may be a newer firmware that is currently shipping on cameras that I have not tested against. However, I am not in position to update the library to support the latest version. I'll likely get back to that in late April or early May. In the meantime, you should use a supported pair as advised above.

RamonGomez, 2017-03-08:
My bad, didn't know i could get older firmware version for the cam. Totally working now with 1.6.2038.

Thanks again.

closed 2017-03-08 by RamonGomez