ifm/libo3d3xx #83: Behavior of framegrabber when using software trigger

cfreundl, 2017-04-20:
The example program below configures the active application on the camera to software trigger, puts it in run mode and issues a single trigger after having created a `FrameGrabber` object for the camera. After a short pause it tries to fetch the result images via the `WaitForFrame()` method, however, it always returns with a timeout. Actually I would expect the method to return immediately with the previously received result. Or is that the intended behavior? 

```c++
#include <o3d3xx_camera.h>
#include <o3d3xx_framegrabber.h>
#include <o3d3xx_image.h>
#include <iostream>

int main(int, char**) {
    o3d3xx::Logging::Init();

    auto cam = std::make_shared<o3d3xx::Camera>();

    // Configure software trigger
    cam->RequestSession();
    cam->SetOperatingMode(o3d3xx::Camera::operating_mode::EDIT);

    auto devCfg = cam->GetDeviceConfig();
    int applicationIdx = devCfg->ActiveApplication();
    cam->EditApplication(applicationIdx);

    auto appCfg = cam->GetAppConfig();
    appCfg->SetTriggerMode(2);

    cam->SetAppConfig(appCfg.get());
    cam->SaveApp();
    cam->CancelSession();

    // Create frame grabber
    auto fg = std::make_shared<o3d3xx::FrameGrabber>(cam, o3d3xx::IMG_CART);

    // Single trigger
    fg->SWTrigger();

    // Sleep long enough to make sure that the result has been sent
    sleep(5);

    // Fetch result
    auto im = std::make_shared<o3d3xx::ImageBuffer>();
    bool ret = fg->WaitForFrame(im.get(), 1000);
    if(!ret) {
        std::cerr << "Timeout waiting for frame" << std::endl;
    }

    return 0;
}
```

tpanzarella, 2017-04-20:
I would do two things to fix this:

1) Instantiate your image buffer above with the frame grabber.
2) Get rid of the `sleep(5)`.

You will then get the frame. So, the method truly is designed to `WaitForFrame` not `GetLastFrame`. In other words, it does not hold on to the last frame in memory and serve that up. I had contemplated that in the past but hadn't come up as an issue. Your thoughts?

tpanzarella, 2017-04-20:
@cfreundl Another pattern to use here if you are concerned with a "race condition" is demonstrated in [this test](https://github.com/lovepark/libo3d3xx/blob/master/modules/oem/test/o3d3xx-oem-framegrabber-tests.cpp#L71) Basically, you can employ a `future` that is already waiting for the frame data then send the trigger.

cfreundl, 2017-04-20:
Of course I did put the `sleep(5)` call in here to make the timing obvious ;-)

I was just thinking that there is no absolutely reliable way to use a single `FrameGrabber` object when the application is configured for software trigger but now that you are mentioning the use of futures I see that it is possible. Maybe we should add an example program on how to use the library for software triggered applications.

It is fine that `WaitForFrame()` always returns the next frame. A customer mentioned that there are situations where he would be actually interested in a "get last frame" semantics, maybe it is worth thinking about adding a method for that to the framegrabber.

cfreundl, 2017-04-20:
As I just raised this issue to make sure that this not an unintended behavior, we can close it again.

closed 2017-04-20 by cfreundl