ifm/libo3d3xx #84: .ssh keys

mikeNike9, 2017-04-24:
Tom

WE have seccussfully deployed OEM partition and can see it in the SWU file.
When we try to ssh into it the camera with the private/public keys pairs we get back an error and cannot connect.
error is 
"Disconnected:No supported authentication metjhods available (server send:publickey)"

Figured I would reach out to you so I can help my programmer tomorrow with this.

mike

tpanzarella, 2017-04-24:
Be sure that the private key has the proper permissions, I believe `0600` is what you want.

mikeNike9, 2017-04-24:
Yes I just used chmod 0600 and this still rendered the same error after redploying.

tpanzarella, 2017-04-24:
I think @graugans can help here too.

graugans, 2017-04-28:
@tpanzarella we are already working on a solution. The issue was @mikeNike9 used the latest official release which does not support the OEM feature. I think we can close this issue

closed 2017-04-28 by tpanzarella