ifm/libo3d3xx #85: oem build

mikeNike9, 2017-04-28:
I am trying to build the oem module.
I am getting this

~/libo3d3xx/modules/oem/build$ cmake -DCMAKE_INSTALL_PREFIX=/opt/oem/usr -DCPACK_INSTALL_PREFIX=/opt/oem/usr -DCMAKE_TOOLCHAIN_FILE=../../../cmake/toolchains/o3d303.cmake ..
CMake Error at /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find Threads (missing: Threads_FOUND)
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.5/Modules/FindThreads.cmake:223 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:63 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/hayden/libo3d3xx/modules/oem/build/CMakeFiles/CMakeOutput.log".
See also "/home/hayden/libo3d3xx/modules/oem/build/CMakeFiles/CMakeError.log".

i can post CMakeError.log and CMakeOutput.log if needed

thoughts

tpanzarella, 2017-04-28:
Did you source in the environment file from the Poky SDK prior to running `cmake`?

mikeNike9, 2017-04-28:
Uhhhh noooo :( ..........

its working now 
thank you and TGIF

closed 2017-04-28 by mikeNike9