ifm/libo3d3xx #86: oem framegrabber

mikeNike9, 2017-04-28:
This is happening when I run the tests?
the BasicFramegrabbing is timing out.

oem@O3D3XX-40-4C-D2:/tmp$ ./o3d3xx-oem-tests
[==========] Running 2 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 2 tests from OEM
[ RUN      ] OEM.BasicFramegrabbing
Timeout while receiving data
/home/hayden/libo3d3xx/modules/oem/test/o3d3xx-oem-framegrabber-tests.cpp:22: Failure
Value of: fg->WaitForFrame(im.get(), 1000)
  Actual: false
Expected: true
Timeout while receiving data
/home/hayden/libo3d3xx/modules/oem/test/o3d3xx-oem-framegrabber-tests.cpp:22: Failure
Value of: fg->WaitForFrame(im.get(), 1000)
  Actual: false
Expected: true
Timeout while receiving data
/home/hayden/libo3d3xx/modules/oem/test/o3d3xx-oem-framegrabber-tests.cpp:22: Failure
Value of: fg->WaitForFrame(im.get(), 1000)
  Actual: false
Expected: true
Timeout while receiving data
/home/hayden/libo3d3xx/modules/oem/test/o3d3xx-oem-framegrabber-tests.cpp:22: Failure
Value of: fg->WaitForFrame(im.get(), 1000)
  Actual: false
Expected: true
Timeout while receiving data
/home/hayden/libo3d3xx/modules/oem/test/o3d3xx-oem-framegrabber-tests.cpp:22: Failure
Value of: fg->WaitForFrame(im.get(), 1000)
  Actual: false
Expected: true
Timeout while receiving data
/home/hayden/libo3d3xx/modules/oem/test/o3d3xx-oem-framegrabber-tests.cpp:22: Failure
Value of: fg->WaitForFrame(im.get(), 1000)
  Actual: false
Expected: true
Timeout while receiving data
/home/hayden/libo3d3xx/modules/oem/test/o3d3xx-oem-framegrabber-tests.cpp:22: Failure
Value of: fg->WaitForFrame(im.get(), 1000)
  Actual: false
Expected: true
Timeout while receiving data
/home/hayden/libo3d3xx/modules/oem/test/o3d3xx-oem-framegrabber-tests.cpp:22: Failure
Value of: fg->WaitForFrame(im.get(), 1000)
  Actual: false
Expected: true
Timeout while receiving data
/home/hayden/libo3d3xx/modules/oem/test/o3d3xx-oem-framegrabber-tests.cpp:22: Failure
Value of: fg->WaitForFrame(im.get(), 1000)
  Actual: false
Expected: true
Timeout while receiving data
/home/hayden/libo3d3xx/modules/oem/test/o3d3xx-oem-framegrabber-tests.cpp:22: Failure
Value of: fg->WaitForFrame(im.get(), 1000)
  Actual: false
Expected: true
 Dispatcher returned with error code: 0
[  FAILED  ] OEM.BasicFramegrabbing (10019 ms)
[ RUN      ] OEM.PixelValidity
Timeout while receiving data
/home/hayden/libo3d3xx/modules/oem/test/o3d3xx-oem-framegrabber-tests.cpp:92: Failure
Value of: fut_tcp.get()
  Actual: false
Expected: true
Timeout while receiving data
 Dispatcher returned with error code: 0
[  FAILED  ] OEM.PixelValidity (9233 ms)
[----------] 2 tests from OEM (19254 ms total)

[----------] Global test environment tear-down
[==========] 2 tests from 1 test case ran. (19255 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 2 tests, listed below:
[  FAILED  ] OEM.BasicFramegrabbing
[  FAILED  ] OEM.PixelValidity

 2 FAILED TESTS
  

thoughts?

tpanzarella, 2017-04-28:
Is your camera set to S/W Trigger? That could cause it.

mikeNike9, 2017-04-28:
i had it set to process interface but it was working like that early today.
I can take it out of process interface though

tpanzarella, 2017-04-28:
For testing purposes, let it free run. It was probably working before because you were testing off-board and the unit tests there explicitly set your camera to factory defaults which would have had the camera streaming.

mikeNike9, 2017-04-28:
I got it to work. Thank you.

closed 2017-05-05 by tpanzarella