ifm/libo3d3xx #88: New to ubuntu, and lost in the basics

R-L-M, 2017-05-05:
I want to install all of the prerequisites, and am not sure how or where to install them.  Could someone get me through the basics?  

Some questions include: 
Do I need ROS or is this a stand-alone software?  It is mentioned, but it says nowhere if it's needed or not. 
Do all of my prerequisites need to be in the same folder?
What is the best process for installing each?

graugans, 2017-05-08:
@R-L-M this is the stand-alone part. There is another project which contains the ROS driver. The ROS driver depends on this project.

To provide help we need more information from you. For example what exact version of Ubuntu are you running. And what are the blocking points for you at the moment. If you are not familiar with C++ then you may consider to use the [Python](https://github.com/cfreundl/o3d3xx-python) or [Ruby](https://github.com/graugans/ruby-o3d3xx) version.

For Ubuntu 16.04 you'll need at least the following packages:
```
sudo apt-get install libboost-all-dev libgtest-dev libgoogle-glog-dev libxmlrpc-c++8-dev build-essential cmake
```

R-L-M, 2017-05-10:
I have Ubuntu 16.04.
My blocking point is as follows:
I have done what I can to install prerequisites, but as I'm new to Ubuntu, I doubt that they're all done, and also doubt that they're all in the right places.   
I have followed steps from the Full Instructions, changing directory to modules/camera, then making build, then changing directory to that new folder, then when I run the next instruction, "cmake -DCMAKE_INSTALL_PREFIX=/usr .." I get errors.  The most glaring I saw from scrolling back up was it complaining that "xmlrpc was not found".

[Errors.tar.gz](https://github.com/lovepark/libo3d3xx/files/990782/Errors.tar.gz)

R-L-M, 2017-05-11:
I post this because there usually aren't notifications from edited posts, and I'm hoping to break through to the actual code writing soon :D

graugans, 2017-05-12:
Please check the front page for assistance how to build without the viewer application. For the unit test the camera needs to be connected.

R-L-M, 2017-05-13:
I'll keep looking at the one spot on the front page that mentions viewer application and try to glean something.  So far no dice in connecting dots.
For the unit test, the failures were much more numerous until I connected the camera.  That was after.
Thanks for your guidance.

tpanzarella, 2017-05-13:
What is the specific `cmake` command line you are running when you try to build the `image` module? To disable the viewer, you would do something like: (assuming you are in the root of the image module)

```
$ mkdir build
$ cd build
$ cmake -DCMAKE_INSTALL_PREFIX=/usr -DBUILD_EXE_VIEWER=OFF ..
$ make
$ make check
$ make package
$ sudo dpkg -i libo3d3xx-image_0.4.9_ubuntu-16.04.2_amd64.deb
```

R-L-M, 2017-05-15:
That worked well.  it had a failure on exposure time, but that's all, and all the commands went off successfully.  
I was using the command $ cmake -DCMAKE_INSTALL_PREFIX=/usr .. as per the instructions when the errors occurred.  
I'm curious as to why I need to disable the image module.  Is there something I can to do include it?

tpanzarella, 2017-05-15:
This does not disable the image module. It simply disables the standalone application: `o3d3xx-viewer`. To include it, you just have a steeper level of prerequisities, including the entire "visualization" component of PCL and its downstream dependencies.

closed 2017-08-25 by tpanzarella

mikeNike9, 2017-09-28:
Have a new minnow board im playing around with.
But the newest wont build right now and I believe its just the viewer.
I have never been able to make the viewer work and would like too.

[ 25%] Built target O3D3XX_CAMERA
[ 50%] Built target O3D3XX_FRAMEGRABBER
[ 75%] Built target O3D3XX_PCICCLIENT
[ 77%] Performing build step for 'O3D3XX_IMAGE'
[  7%] Built target o3d3xx_image
[ 14%] Built target o3d3xx_image_static
[ 21%] Built target o3d3xx_image_shared
[ 28%] Automatic moc for target o3d3xx-viewer
[ 28%] Built target o3d3xx-viewer_automoc
make[5]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libproj.so', needed by 'src/bin/o3d3xx-viewer'.  Stop.
CMakeFiles/Makefile2:241: recipe for target 'src/bin/CMakeFiles/o3d3xx-viewer.dir/all' failed
make[4]: *** [src/bin/CMakeFiles/o3d3xx-viewer.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMakeFiles/O3D3XX_IMAGE.dir/build.make:116: recipe for target 'O3D3XX_IMAGE-prefix/src/O3D3XX_IMAGE-stamp/O3D3XX_IMAGE-build' failed
make[2]: *** [O3D3XX_IMAGE-prefix/src/O3D3XX_IMAGE-stamp/O3D3XX_IMAGE-build] Error 2
CMakeFiles/Makefile2:179: recipe for target 'CMakeFiles/O3D3XX_IMAGE.dir/all' failed
make[1]: *** [CMakeFiles/O3D3XX_IMAGE.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

mikeNike9, 2017-09-28:
If anything I would like to figure out why I cant finishing building.
i tried the above by turning off the viewer but I still get this same error.

mikeNike9, 2017-09-28:
months back I had to revert back to a later release and revert back to a different firmware on my ifm to allow the build to finish. 
Let me know what you guys think 
thank you