ifm/libo3d3xx #94: EnableRectificationAmplitudeImage

mikeNike9, 2017-06-01:
I have tried to use command line tools and cannot successfully Set the "EnableRectificationAmplitudeImage":"true"

Any tips on configuring the camera.json

tpanzarella, 2017-06-01:
Could you provide either as an attachment or inline the exact JSON you are using and exactly how you are calling the command line tool?

mikeNike9, 2017-06-01:
echo '{"o3d3xx":{"Apps":[{"Index":"2","Imager":{"EnableRectificationAmplitudeImage": "true", "EnableRectificationDistanceImage":"true"}}],"Device":{ActiveApplication":"1"}}}' | o3d3xx-config

tpanzarella, 2017-06-01:
Please also paste in the output of o3d3xx-dump so that I can see the state of the camera prior to trying to piping in this JSON.

tpanzarella, 2017-06-02:
@mikeNike9 The problem you are having is that you have a syntax error in your json. You are missing the opening quote around the word `ActiveApplication` in the `Device` section. For example, on a camera that I just factory reset, I can run this:

```
$ echo '{"o3d3xx":{"Apps":[{"Index":"2","Imager":{"EnableRectificationAmplitudeImage": "true", "EnableRectificationDistanceImage":"true"}}],"Device":{"ActiveApplication":"1"}}}' | o3d3xx-config
```
Now, as you know (I assume), you are turning on image rectification on the "second application" (i.e., `"Index":"2"`). However, you are marking active, the "first application" (`"ActiveApplication":"1"`). So, in my scenario (I don't know what yours is because I do not have a dump of your camera state), with the camera initially in a factory reset mode (I have one application, at Index = 1, and it is marked as active), after I run your (corrected) JSON through `o3d3xx-config`, I get two applications on my camera (since an application at Index = 2 does not exist, `o3d3xx-config` assumes you want to create it as a new one). It is the second application that has rectification turned on, and (in my case) the first application (Index = 1) is active but does *not* have rectification turned on.

To elucidate this, after piping in your (corrected) JSON, I now have two applications and the first one is marked active (i.e., `*`):

```
$ o3d3xx-ls
* [1] id=476707713, name=Sample Application, description=
  [2] id=1149375325, name=New application, description=
```
However, as your JSON requests, the new application (at Index = 2) has the rectification flags turned on. A dump just after piping in your (corrected) JSON into a factory reset O3D303 is attached for your reference. And for convenience to simply check the rectification flags....

The application at Index = 1:

```
$ o3d3xx-dump | jq .o3d3xx.Apps[0].Imager | grep -i 'Rectification'
  "EnableRectificationAmplitudeImage": "false",
  "EnableRectificationDistanceImage": "false",
```
And at Index = 2:

```
$ o3d3xx-dump | jq .o3d3xx.Apps[1].Imager | grep -i 'Rectification'
  "EnableRectificationAmplitudeImage": "true",
  "EnableRectificationDistanceImage": "true",
```

[camera.zip](https://github.com/lovepark/libo3d3xx/files/1046334/camera.zip)

mikeNike9, 2017-06-02:
Sorry for late response.
I am going to factory reset a camera and try to see if this new syntax will work for my scenario.
Everything you are saying makes complete sense.
I will probably be sending another request later once I try to put a final application on the camera to run on start up of the camera. 
I have had alot of success embedding all of our 3D measurement classes and algorithms on the the 303.
The cool thing is that the camera is able to run the filters extremely fast and does not run out of memory ever.

mikeNike9, 2017-06-02:
Thank you tom.
You were correct about the Index problem and the syntax error on active application
Everything is well now.

closed 2017-06-02 by mikeNike9