ifm/libo3d3xx #98: Startup

mikeNike9, 2017-06-13:
Tom 

Was wondering where in the OEM partition do I need to put my exe to have it startup when the sensor is rebooted or powered on?

tpanzarella, 2017-06-13:
You should place a standard SysV init script in `/opt/oem/etc/init.d` that will wrap and start your executable. All scripts in that directory are started in lexographical order, so the typical way to impose an ordering is to name things like: `00_foo`, `01_bar`, `99_baz` etc. There is no inter-dependency system (like `systemd` or other such init-like frameworks) between services, you need to *intelligently* manage that yourself. HTH.

tpanzarella, 2017-06-13:
I should also mention that this is really unrelated to `libo3d3xx`. It is more of a question for the ifm firmware guys, but, the above should get you going.

closed 2017-06-13 by tpanzarella