ifm/libo3d3xx #99: Triggering multiple cameras from oem partition

mikeNike9, 2017-06-16:
Tom 

Can I use the same Frame Grabber on the Camera with multiple camera objects?
I have been trying a few things today but have not succeeded. I will try and add a new  IP address but it will continue to grab the default camera IP address when I call the the FrameGrabber.
auto cam = std::make_shared<o3d3xx::Camera>(); 
auto cam2= std::make_shared<o3d3xx::Camera>("192.168.0.71"); 
auto fg = std::make_shared<o3d3xx::oem::FrameGrabber>(cam2, o3d3xx::IMG_AMP|o3d3xx::IMG_RDIS|o3d3xx::IMG_CART);
  auto im = std::make_shared<o3d3xx::oem::ImageBuffer>();
      EXPECT_TRUE(fg->WaitForFrame(im.get(), 1000));
      pcl::PointCloud<o3d3xx::PointT>::Ptr cloud = im->Cloud();

This will return the point cloud of cam and not of the cam2 (192.168.0.71)

I have other ways grab the points off the other cameras but figured I would try and grab a frame for fun first.

tpanzarella, 2017-06-16:
Mike,

The OEM framegrabber is a "zero copy" framegrabber. I.e., it returns to you data directly from the internal imager buffers -- it does not get its data from the PCIC server on the device. If you want to fetch data from a remote camera from code running on another camera, you need to use the standard libo3d3xx TCP framegrabber -- I.e., at that point you are building a distributed system and a network connection is required. 

Tom Panzarella
Founder & CEO
Love Park Robotics, LLC
2025 Washington Ave, Suite 217
Philadelphia, PA 19146 USA

mobile: +1 215 341 1754
tom@loveparkrobotics.com
http://loveparkrobotics.com

graugans, 2017-06-29:
@mikeNike9 you should either cross compile the standard framegrabber like @tpanzarella said or implement an onboard camera daemon which is managing this for you. In the second approach you may can reduce the data size my some kind of pre processing on each node.

tpanzarella, 2017-06-29:
Christian, is the standard TCP framegrabber included on the OEM images still?

graugans, 2017-06-30:
@tpanzarella no, there was no reason because of the zero-copy approach.

tpanzarella, 2017-06-30:
Two use cases I can think of include:

1. Multi camera systems
2. One code base that cross-compiles (and runs) to on or off-board deployment.

closed 2017-08-25 by tpanzarella