ifm/libo3d3xx #100: Cant ping the camera

Eirtrand, 2017-07-04:
I'm running Ubuntu 16.04 together with the O3D303 camera. Its connected to the ethernet port on my laptop. It shows up as "wired connection 1"
 A ping to 192.168.0.69 returns:
From 217 to 17.212.193 icmp_seq=4 Packet filtered
From 217 to 17.212.193 icmp_seq=5 Packet filtered
From 217 to 17.212.193 icmp_seq=6 Packet filtered

It gets the hardware address 28:D2:44:F9:33:29
My ip is 192.168.1.74 The wireshark output when connecting to the network can be seen here:
[image]

With that information I would like to ask what I need to do in order to find the camera. How is the network going to be configured? Can I be connected to wifi at the same time? Is something suppose to show up on http://192.168.0.69/api/rpc/v1/com.ifm.efector/ in the browser?

graugans, 2017-07-04:
@Eirtrand The camera does provide a [link local ip](https://en.wikipedia.org/wiki/Link-local_address) to be accessed by machines configured as DHCP and connected to the camera. In this case your machine can not get an IPv4 address from the DHCP server because it is connected to the camera on this port. Windows default behaviour is to fall back to an link local ip after a couple of retries. Frankly I have no idea what is the default behavior on Ubuntu.  Anyway, on Windows you can check your Network settings by issuing the following command after spawning a windows ``cmd.exe`` Shell

```
ipconfig -a
```

On Ubuntu you can check your network configuration with ``/sbin/ifconfig``

```
eno1      Link encap:Ethernet  HWaddr 6c:0b:84:XX:XX:XX  
          inet addr:172.XX.XX.XX  Bcast:172.XX.XX.XXX  Mask:255.255.255.0
          inet6 addr: fe80:::0/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:395874 errors:0 dropped:0 overruns:0 frame:0
          TX packets:162340 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:241198884 (241.1 MB)  TX bytes:37523770 (37.5 MB)
          Interrupt:20 Memory:fb100000-fb120000 

enx00249b1e3892 Link encap:Ethernet  HWaddr 00:24:9b:XX:XX:XX  
          inet addr:192.168.0.201  Bcast:192.168.0.255  Mask:255.255.255.0
          inet6 addr: fe80::0/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:61223 errors:0 dropped:0 overruns:0 frame:0
          TX packets:143721 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:3558872 (3.5 MB)  TX bytes:215354625 (215.3 MB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:202499 errors:0 dropped:0 overruns:0 frame:0
          TX packets:202499 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:334947494 (334.9 MB)  TX bytes:334947494 (334.9 MB)
```
To access the camera your PC and the camera need to be in the same network segment.

Eirtrand, 2017-07-04:
Thank you for the response. The problem was solved by connecting the camera to my router. However I now faced another problem. I can not update the firmware on the camera, it says that I need a static IP. I turned of DHCP on the camera and also made a static ip for my computer. Whit this done the error is still there, asking me to change the device to a static ip address

closed 2017-07-04 by Eirtrand

graugans, 2017-07-04:
Could you please be so kind and send the output of the ip config utily for your Operating system from my comment above?

Eirtrand, 2017-07-04:
I solved it by creating a network on the same subnet as the default address. Because it didn't work with the temporary address that the camera was given.  I now have everything working with ros :) This might me be pure logic, but it still took me a while to figure out!

graugans, 2017-07-05:
@Eirtrand If there are any hints how to improve the documentation I would be more than happy to improve it.