ifm/libo3d3xx #102: Invalid unit-vector from the O3D sensor at 0, 0

bc-jmoork, 2017-08-07:
For some reason the first unit-vector for our latest O3D camera (with larger FOV) is invalid.

unit_vector[0, 0, :] = array([  0.00000000e+00,   7.21781871e+14,   1.04866115e-28]

The value of e+14 is impossible and should not be present in the unit-vector. Our current fix or hack is to set the unit-vector for [0, 0] to the value in [0, 1]

HW info from o3d3xx-dump is shown below:

        "HWInfo":
        {
            "Mainboard": "#!02_M381_003_02888804_008022954",
            "Illumination": "#!02_I300_001_02895643_008001175",
            "MiraSerial": "1070-5908-00b4-08d0",
            "MACAddress": "00:02:01:40:65:12",
            "Frontend": "#!02_F342_C34_17_00012_008023607",
            "Diagnose": "#!02_D322_C32_02879150_008023267",
            "Connector": "#!02_A300_001_02870331_008001175"
        }

graugans, 2017-08-08:
@bc-jmoork could you please be so kind and provide the software version from the dump?

```json
        "SWVersion": {
            "Main_Application": "0.4.1284",
            "Linux": "Linux version 3.14.34-rt31-yocto-standard-00014-gaefa18d-dirty (jenkins@dettlx152) (gcc version 4.9.2 (GCC) ) #1 SMP PREEMPT RT Wed Aug 2 10:15:20 CEST 2017",
            "IFM_Software": "1.8.1330",
            "Diagnostic_Controller": "0.4.1284",
            "Calibration_Version": "0.0.0",
            "ELDK": "GOLDENEYE_YOCTO_HARDFP-351-06d9c894636352a6c93711c7284d02b0c794a527",
            "Calibration_Device": "ff:ff:ff:ff:ff:ff",
            "Algorithm_Version": "2.0.5"
        },
```

bc-jmoork, 2017-08-08:
Hi @graugans 
{
  "Version": "1.8.768",
  "Article": ""
}

bc-jmoork, 2017-08-09:
@graugans Any update on this?

graugans, 2017-08-10:
@bc-jmoork could you please be so kind and send me the full dump to christian.ege (at) ifm.com this empty article number looks a bit weired to me.

bc-jmoork, 2017-08-23:
@graugans I updated the libo3d3xx from 1.42 to 1.50 and we are no longer seeing the issue with the unit vector. Thanks a lot for your assistance.

tpanzarella, 2017-08-23:
@bc-jmoork I do not understand the version numbers you are citing. Where do you get 1.42 and 1.50?

bc-jmoork, 2017-08-23:
@tpanzarella, I am sorry, we updated from v.0.42 to v.0.50.

tpanzarella, 2017-08-23:
Always be sure to consult the main README file at the top-level of this source distribution. It maintains a software compatibility matrix that maps versions of libo3d3xx to versions of the ifm firmware that have been tested to work properly. This table is rendered nicely at the source URL of this github repo.

Specific to this issue, I think the "fix" that corrected this was when libo3d3xx added support for the ifm image chunk headers version 2. Specifically, this support was added in `0.4.9`. The [ChangeLog](https://github.com/lovepark/libo3d3xx/blob/master/ChangeLog.md) has more details.

closed 2017-08-25 by tpanzarella