ifm/libo3d3xx #103: THIS IS A QUESTION ON WINDOWS

mikeNike9, 2017-08-28:
Hello all

I have had some issues with AC power failure in some locations.
This AC power failure will sometimes freeze the cameras.
I will be able to connect to the camera successfully but when the camera is updated via pmdUpdate(); 
the windows sdk gets tied up and will not return the object PMD_OK.

I fix this issue by rebooting the camera.

I would like to create a script to reboot all of my cameras in my system when this situation occurs so the system will fix itself rather than either me rebooting via the vision assistant or physically powering it down for 15-30 seconds.

I know there are command line tools for linux but I was aware of any in Windows.
I have a note into Germany but wanted to reach out here on the forum and see if you any of you guys had any ideas.

thank you 

mike Jr

FreightSnap LLC

graugans, 2018-01-25:
Hello @mikeNike9,
We have recently added [Microsoft Windows support](https://github.com/lovepark/ifm3d/blob/master/doc/windows.md) for the camera module on the [ifm3d](https://github.com/lovepark/ifm3d) library which is the successor of the libo3d3xx. Discussion and feedback about this feature takes place in this issue [#21](https://github.com/lovepark/ifm3d/issues/21) any feedback is appreciated.

mikeNike9, 2018-01-25:
I see!!! Im excited to test it out!

closed 2020-07-27 by theseankelly