ifm/libo3d3xx #104: camera module on windows

Postea-CS, 2017-08-28:
Hi 
I am wondering if i only need the camera module,  are there any issues with compilation the code on windows?

cheers,
CS

graugans, 2017-08-29:
Hi @Postea-CS we are currently trying to port the ``ifm3d`` (successor of ``libo3d3xx``) to windows. It looks like the hardest part is getting the ``xmlrpc-c`` compiled. If you are interested I can share my xmlrpc-c CMake stuff I have at the moment.

graugans, 2017-09-06:
@Postea-CS I have some good news for you. Today I released a Version of [xmlrpc-c](https://github.com/graugans/xmlrpc-c) which is capable to be build with the latest version of CMake on Visual Studio 2017 Community Edition. I requires a *libcurl* as transport layer. If there are any questions, feel free to create an issue on this project.

graugans, 2017-11-06:
# :exclamation: This is a preliminary documentation how to build the [ifm3d](https://github.com/lovepark/ifm3d) on Windows. Most concepts should apply to the libo3d3xx as-well

## How to build ifm3d library from source on Windows.

### Prepare dependent tools

Make sure that you have installed CMake and VS2017 64bit before you move to the next step.

* [CMake v3.9.1](http://www.cmake.org)
* [Visual Studio Community 2017](https://www.visualstudio.com)

### Download dependencies

Start with downloading Glog, Gtest and XMLRPC libraries from repository listed below to ``3rdParty`` folder.

* [Gtest](https://github.com/google/googletest)
* [Glog](https://github.com/google/glog)
* [xmlrpc-c](https://github.com/graugans/xmlrpc-c) This is a special version with CMake support. Follow the instructions how to build in this [Readme](https://github.com/graugans/xmlrpc-c/blob/1.33.14-cmake/README.md)
* [curl](https://github.com/curl/curl) For the [xmlrpc-c library](https://github.com/graugans/xmlrpc-c/blob/1.33.14-cmake/README.md#prerequisites)

### Install dependencies

Install the necessary libraries on your system. In the following example we assume ``D:\3rdparty`` is used as a base path.

* [Boost v1.64.0](http://www.boost.org)
* [PCL v1.8.1](https://github.com/PointCloudLibrary/pcl/releases)
* [OpenCV v3.3](http://opencv.org)

In case of using Visual Studio 15 2017 we need to adapt "opencv/build/OpenCVConfig.cmake" file.

Replace:
```
    elseif(MSVC_VERSION EQUAL 1910 OR MSVC_VERSION EQUAL 1911)
        set(OpenCV_RUNTIME vc15)
```
With:
```
    elseif(MSVC_VERSION EQUAL 1910 OR MSVC_VERSION EQUAL 1911)
        set(OpenCV_RUNTIME vc14)
```

After instalation you need to set paths to each of three libraries in ~~"ifm3d/CMakeLists.txt"~~ file.

```
# Where to find Boost path
    set(BOOST_INCLUDEDIR    "D:/locallibs/boost_1_64_0/")
    set(BOOST_LIBRARYDIR    "D:/locallibs/boost_1_64_0/lib64-msvc-14.1")
    set(BOOST_ROOT          "D:/locallibs/boost_1_64_0/boost/")
# Where to find OpenCV path
    set(OpenCV_DIR          "D:/locallibs/opencv/build/")
# Where to find PCL path
    set(PCL_DIR             "D:/locallibs/pcl1.8.1/cmake/")
```

### Generate VisualStudio project via CMake.

Due to the fact the Visual Studio compiler needs some command line variables correct initialized. It is recommended to use the the Visual Studio command line.

```
$ mkdir build
$ cd build
$ cmake ..\ifm3d -G "Visual Studio 15 2017 Win64" -DBUILD_SHARED_LIBS=ON -DMUST_BUILD_CURL_CLIENT=ON -DBUILD_TESTING=OFF
```    

### Build projects in Visual Studio 2017 64bit.

You will find IFM3D.sln in build folder, created in previous step, launch it.

---

:exclamation: For the libo3d3xx library it is recommended to build the modules separately and not with the top-level CMake file.

graugans, 2018-01-25:
We have recently added [Microsoft Windows support](https://github.com/lovepark/ifm3d/blob/master/doc/windows.md) for the camera module on the [ifm3d](https://github.com/lovepark/ifm3d) library which is the successor of the libo3d3xx. Discussion and feedback about this feature takes place in this issue [#21](https://github.com/lovepark/ifm3d/issues/21) any feedback is appreciated.

closed 2020-07-27 by theseankelly