ifm/libo3d3xx #105: XMLRPC was not found

ARICKERT0003, 2017-09-21:
Ubuntu 16.04
CMake version 3.9.3

I have followed the given instructions
for all of the following I am root

cd address/to/libo3d3xx-master
mkdir build
cd build
cmake ..

I have downloaded all the dependencies for libo3d3xx but I get the following error: 

CMake Error at address/to/cmake-3.9.3-Linux-x86_64/share/cmake-3.9/Modules/FindPackageHandlesStandardsArgs.cmake:137 (message):
XMLRPC was not found. Make sure the entries XMLRPC_* are set. (missing: XMLRPC_C_FOUND XMLRPC_LIBRARIES)

Call Stack (most recent call first):
address/to/cmake-3.9.3-Linux-x86_64/share/cmake-3.9/Modules/Find_Package_Handle_Standards_Args)
CMakeLists.txt:377 (_FPHSA_FAILURE_MESSAGE)

address/to/cmake-3.9.3-Linux-x86_64/share/cmake-3.9/Modules/FindPackageHandlesStandardsArgs.cmake:86 (find_package)

Here is a picture of all the libxmlrpc's I have
[image]

Could anyone help me fix this issue? I have searched for similar error but have not found any.
Thank you

tpanzarella, 2017-09-22:
I am not familiar with the GUI you show above and so am not sure how to read it. However, to get what you need regarding XMLRPC to build `libo3d3xx` if you do this, you should get what you need:

```
    $ sudo apt-get install libxmlrpc-c++8-dev
```

Unrelated to your question, if you are just getting started with the O3D, unless you have specialized concerns (there are a few), I'd suggest you consider using the [ifm3d](https://github.com/lovepark/ifm3d) library as this is the successor to `libo3d3xx`.  FWIW, `ifm3d` also relied on the same xmlprc client library so, what I stated above will apply there too.

HTH.

graugans, 2017-09-25:
Hello @ARICKERT0003 there is a list of packages needed to get started for libo3d3xx on Ubuntu 16.04 in the issue #88 As @tpanzarella already stated it might be a good choice to use ifm3d instead of libo3d3xx. If you are planning to use ROS with the cameras I suggest to install ROS before this driver otherwise you may end up with conflicting OpenCV versions.

ARICKERT0003, 2017-09-28:
This error was resolved.  It said it was a XMLRPC error, however libcurl was at fault.  libcurl.so was not linked in /usr/lib/x86_64-linux-gnu/ to libcurl-gnutls.so.  
This was found by looking at the advanced options in ccmake under XMLRPC_curl_LIBRARY during the build process.
Thank you for the responses.

closed 2017-09-28 by ARICKERT0003