ifm/libo3d3xx #108: ssh permission denied

asantrach, 2017-10-09:
I have created and deployed an oem partition to my camera but when I try to ssh to it I get a Permission denied(publickey) error. I've update the permissions and that doesn't seem to work. Any ideas around this error? Does it matter which firmware I am making a partition of?

cfreundl, 2017-10-12:
Can you provide the commandline and the output of a login attempt with the `-v` option added to the ssh call, assuming you are using ssh in a console?

asantrach, 2017-10-12:
ssh 10.10.10.81 -v

output:
OpenSSH_6.6.1, OpenSSL 1.0.1f 6 Jan 2014
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 19: Applying options for *
debug1: Connecting to 10.10.10.81 [10.10.10.81] port 22.
debug1: Connection established.
debug1: identity file /home/amelia/.ssh/id_rsa type 1
debug1: identity file /home/amelia/.ssh/id_rsa-cert type -1
debug1: identity file /home/amelia/.ssh/id_dsa type -1
debug1: identity file /home/amelia/.ssh/id_dsa-cert type -1
debug1: identity file /home/amelia/.ssh/id_ecdsa type -1
debug1: identity file /home/amelia/.ssh/id_ecdsa-cert type -1
debug1: identity file /home/amelia/.ssh/id_ed25519 type -1
debug1: identity file /home/amelia/.ssh/id_ed25519-cert type -1
debug1: Enabling compatibility mode for protocol 2.0
debug1: Local version string SSH-2.0-OpenSSH_6.6.1p1 Ubuntu-2ubuntu2.8
debug1: Remote protocol version 2.0, remote software version dropbear_2014.66
debug1: no match: dropbear_2014.66
debug1: SSH2_MSG_KEXINIT sent
debug1: SSH2_MSG_KEXINIT received
debug1: kex: server->client aes128-ctr hmac-md5 none
debug1: kex: client->server aes128-ctr hmac-md5 none
debug1: sending SSH2_MSG_KEX_ECDH_INIT
debug1: expecting SSH2_MSG_KEX_ECDH_REPLY
debug1: Server host key: RSA c6:0a:47:7c:18:f3:79:78:c6:d9:52:ac:0a:7c:68:f5
debug1: Host '10.10.10.81' is known and matches the RSA host key.
debug1: Found key in /home/amelia/.ssh/known_hosts:1
debug1: ssh_rsa_verify: signature correct
debug1: SSH2_MSG_NEWKEYS sent
debug1: expecting SSH2_MSG_NEWKEYS
debug1: SSH2_MSG_NEWKEYS received
debug1: SSH2_MSG_SERVICE_REQUEST sent
debug1: SSH2_MSG_SERVICE_ACCEPT received
debug1: Authentications that can continue: publickey
debug1: Next authentication method: publickey
debug1: Offering RSA public key: /home/amelia/.ssh/id_rsa
debug1: Authentications that can continue: publickey
debug1: Trying private key: /home/amelia/.ssh/id_dsa
debug1: Trying private key: /home/amelia/.ssh/id_ecdsa
debug1: Trying private key: /home/amelia/.ssh/id_ed25519
debug1: No more authentication methods to try.
Permission denied (publickey).

asantrach, 2017-10-12:
I've also tried ssh 10.10.10.81 -i ~/.ssh/id_rsa -v but that results in the same output

cfreundl, 2017-10-13:
My first guess that the login failure is due to a wrong username. Can you make sure that you are logging onto the camera as user `oem`? As the commandline and the ssh output do not show an explicit username, I assume that your local username (`amelia`) is being used.

asantrach, 2017-10-13:
okay when I give it a user oem I get a slightly different error.

ssh oem@10.10.10.81 -i ~/.ssh/id_rsa -v

OpenSSH_6.6.1, OpenSSL 1.0.1f 6 Jan 2014
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 19: Applying options for *
debug1: Connecting to 10.10.10.81 [10.10.10.81] port 22.
debug1: Connection established.
debug1: identity file /home/amelia/.ssh/id_rsa type 1
debug1: identity file /home/amelia/.ssh/id_rsa-cert type -1
debug1: Enabling compatibility mode for protocol 2.0
debug1: Local version string SSH-2.0-OpenSSH_6.6.1p1 Ubuntu-2ubuntu2.8
debug1: Remote protocol version 2.0, remote software version dropbear_2014.66
debug1: no match: dropbear_2014.66
debug1: SSH2_MSG_KEXINIT sent
debug1: SSH2_MSG_KEXINIT received
debug1: kex: server->client aes128-ctr hmac-md5 none
debug1: kex: client->server aes128-ctr hmac-md5 none
debug1: sending SSH2_MSG_KEX_ECDH_INIT
debug1: expecting SSH2_MSG_KEX_ECDH_REPLY
debug1: Server host key: RSA c6:0a:47:7c:18:f3:79:78:c6:d9:52:ac:0a:7c:68:f5
debug1: Host '10.10.10.81' is known and matches the RSA host key.
debug1: Found key in /home/amelia/.ssh/known_hosts:1
debug1: ssh_rsa_verify: signature correct
debug1: SSH2_MSG_NEWKEYS sent
debug1: expecting SSH2_MSG_NEWKEYS
debug1: SSH2_MSG_NEWKEYS received
debug1: SSH2_MSG_SERVICE_REQUEST sent
debug1: SSH2_MSG_SERVICE_ACCEPT received
debug1: Authentications that can continue: publickey
debug1: Next authentication method: publickey
debug1: Offering RSA public key: /home/amelia/.ssh/id_rsa
debug1: Server accepts key: pkalg ssh-rsa blen 279
**Agent admitted failure to sign using the key.**
debug1: No more authentication methods to try.
Permission denied (publickey)

asantrach, 2017-10-13:
with an updated gem I was able to log in using SSH_AUTH_SOCK=0 ssh oem@10.10.10.81

closed 2017-10-13 by asantrach

graugans, 2017-10-23:
@asantrach This ``SSH_AUTH_SOCK=0`` is interesting. What Ubuntu version are you using?

asantrach, 2017-10-23:
14.04 
Adding the private key to the ssh-agent allows you to use ssh normally