ifm/libo3d3xx #109: Setting 100K pixel resolution mode not working

nathangeorge1, 2017-10-18:
Hello,

I tried using the ex-100k.cpp to set the resolution config parameter to 1 so that I can view in 100k mode, but when I use o3d3xx-dump to view the config json, the Resolution is still 0.

I also tried `echo '{"o3d3xx":{"Apps":{"Imager":{"Resolution":"1"}}}}' | o3d3xx-config`, as mentioned in the README; but it didn't work either.

How do I set the resolution to 100k?

Thanks,
Nathan

nathangeorge1, 2017-10-18:
Update: I used this document and the 100k mode is set!

https://github.com/lovepark/libo3d3xx/blob/master/doc/100k.md

There is an update up top that says ` Simply turn it on by setting the Resolution parameter of the JSON serialization of the camera.`; but I think that is misleading as I needed to import a new application with `o3d3xx-ifm-import -f 100k.o3d3xxapp -a` to get it working.

Is this a bug?

tpanzarella, 2017-10-18:
This is not a bug.

closed 2017-10-25 by tpanzarella

nathangeorge1, 2018-01-22:
Hello @tpanzarella, I would like to revisit this issue as there appears to be a misunderstanding, because my original problem wasn't solved. With one of the latest firmwares, trying to set the 100K mode using the following command doesn't work:

`echo '{"o3d3xx":{"Apps":{"Imager":{"Resolution":"1"}}}}' | o3d3xx-config`

Attached is my output from `o3d3xx-dump`: [dump.txt](https://github.com/lovepark/libo3d3xx/files/1653583/dump.txt)

How do I set Resolution via command line?

nathangeorge1, 2018-01-22:
Actually, we upgraded to firmware version 1.6.2114, and by using libo3d3xx/doc/attic/100k.o3d3xxapp, the resolution parameter does change to 1, although using the following command doesn't change is back to normal resolution:

`echo '{"o3d3xx":{"Apps":{"Imager":{"Resolution":"0"}}}}' | o3d3xx-config`

It would still be interesting to learn why the echo command is not working.

graugans, 2018-01-23:
@nathangeorge1 there is a typo in your configuration command. you have created a new application with the resolution set to one. But you wanted to alter the Resolution parameter I guess.

Let's check this:

```
o3d3xx@06d3b9c50eb5:~/libo3d3xx$ o3d3xx-ls 
* [1] id=822604987, name=Application, description=No Description
o3d3xx@06d3b9c50eb5:~/libo3d3xx$ echo '{"o3d3xx":{"Apps":{"Imager":{"Resolution":"1"}}}}' | o3d3xx-config
o3d3xx@06d3b9c50eb5:~/libo3d3xx$ o3d3xx-ls 
* [1] id=822604987, name=Application, description=No Description
  [2] id=1181261689, name=New application, description=
```

The fix is to select the right application, but first get rid of the extra application:

```
o3d3xx@06d3b9c50eb5:~/libo3d3xx$ o3d3xx-rm -i 2  
o3d3xx@06d3b9c50eb5:~/libo3d3xx$ o3d3xx-ls 
* [1] id=822604987, name=Application, description=No Description
```

Now change the resolution:
```
o3d3xx@06d3b9c50eb5:~/libo3d3xx$  echo '{"o3d3xx":{"Apps":[{"Index":"1","Imager":{"Resolution":"1"}}]}}' | o3d3xx-config
o3d3xx@06d3b9c50eb5:~/libo3d3xx$ o3d3xx-ls 
* [1] id=822604987, name=Application, description=No Description
o3d3xx@06d3b9c50eb5:~/libo3d3xx$ o3d3xx-dump | jq '{version: {ifm: .o3d3xx.SWVersion.IFM_Software, lib: .o3d3xx.libo3d3xx}, resolution: .o3d3xx.Apps[0].Imager.Resolution}'
{
  "version": {
    "ifm": "1.6.2114",
    "lib": "701"
  },
  "resolution": "1"
}
```

You can confirm this with an ifm vision assitant running in monitor mode the resolution changes.

nathangeorge1, 2018-01-25:
thanks; will try as soon as I can