ifm/libo3d3xx #112: Cross Compilation with PCL

asantrach, 2017-11-01:
Hello,

I'm trying to compile a simple program to test the PCL and OpenCV libraries but I'm running into some errors. I am using the o3d303.cmake toolchain file and the poky environment. I've sourced environment but when configuring I get a warning regarding openni. 
I shouldn't need openni but the PCLConfig.cmake tries to find it anyway. Do you have a work around for this? Possibly a FindVTK.cmake that can be used in the same way a FindOpenCV.cmake works?

asantrach, 2017-11-01:
I don't know if your fix for VTK errors with -DUBUNTU_VERSION:STRING=16.04 is the way to go, but when I did try it I no longer get a the missing openni warning and instead I get a could not find Threads error. 

CMake Error at /opt/poky/1.8.1/sysroots/armv7ahf-vtp-neon-poky-linux-gnueabi/usr-/share/pcl-1.7/PCLConfig.cmake:38 (message): common is required by boost was not found.

closed 2017-11-03 by asantrach

graugans, 2017-11-06:
@asantrach could you please be so kind and share how you've solved the issue?

asantrach, 2017-11-13:
I had just simply forgotten or accidently deleted the following 
`link_directories(${PCL_LIBRARY_DIRS})`

Once I added that everything compiled. I still get a warning about missing openni but it doesn't seem to be an issue at this point.