ifm/libo3d3xx #115: Depth image is blank when saving usining ex-file_io example

niallomahony93, 2017-12-21:
I am using ex-file_io in the examples folder to save images. The amplitude and point cloud images are ok but the 'radial-distance.png image appears to be just black. I can view all images no problem on rvis but  do not have o3d3xx-viewer setup properly yet as I am using opencv 3.3. libo3d3xx: version=0.7.1. o3d3xx-ros version = 0.2.5. Dump output is: 
{
    "o3d3xx": {
        "libo3d3xx": "701",
        "Date": "Thu Dec 21 14:58:23 2017",
        "HWInfo": {
            "Connector": "#!02_A300_001_02105252_008001175",
            "Diagnose": "#!02_D322_C30_02157009_008023093",
            "MiraSerial": "1630-5392-0022-0134",
            "MACAddress": "00:02:01:40:2F:75",
            "Frontend": "#!02_F342_C32_16_00004_008022797",
            "Mainboard": "#!02_M381_003_02178445_008022954",
            "Illumination": "#!02_I300_001_02159299_008001175"
        },
        "SWVersion": {
            "IFM_Software": "1.6.2114",
            "Main_Application": "0.4.222",
            "Algorithm_Version": "1.2.19",
            "Calibration_Device": "00:02:01:40:2f:75",
            "ELDK": "GOLDENEYE_ELDK-2016-07-14_09-36-02-2e9219cce224332ecbe2f36934d2c3c7aa738c78",
            "Calibration_Version": "0.9.0",
            "Linux": "Linux version 3.11.0-00002-gd1bbe0b (eswadmin@dettlx79) (gcc version 4.7.2 (GCC) ) #1 SMP Fri Jun 24 16:34:26 CEST 2016",
            "Diagnostic_Controller": "0.3.207"
        },
        "Device": {
            "Name": "New sensor",
            "Description": "",
            "ActiveApplication": "1",
            "PcicTcpPort": "50010",
            "PcicProtocolVersion": "3",
            "IOLogicType": "1",
            "IODebouncing": "true",
            "IOExternApplicationSwitch": "0",
            "SessionTimeout": "30",
            "ServiceReportPassedBuffer": "15",
            "ServiceReportFailedBuffer": "15",
            "ExtrinsicCalibTransX": "0",
            "ExtrinsicCalibTransY": "0",
            "ExtrinsicCalibTransZ": "0",
            "ExtrinsicCalibRotX": "0",
            "ExtrinsicCalibRotY": "0",
            "ExtrinsicCalibRotZ": "0",
            "EvaluationFinishedMinHoldTime": "10",
            "SaveRestoreStatsOnApplSwitch": "true",
            "IPAddressConfig": "0",
            "PasswordActivated": "false",
            "OperatingMode": "1",
            "DeviceType": "1:2",
            "ArticleNumber": "O3D313",
            "ArticleStatus": "AB",
            "UpTime": "26.814166666666701",
            "ImageTimestampReference": "0",
            "TemperatureFront1": "3276.6999999999998",
            "TemperatureFront2": "3276.6999999999998",
            "TemperatureIMX6": "38.3289985656738",
            "TemperatureIllu": "39.5",
            "PNIODeviceName": "",
            "EthernetFieldBus": "1",
            "EthernetFieldBusEndianness": "0"
        },
        "Net": {
            "MACAddress": "",
            "NetworkSpeed": "0",
            "StaticIPv4Address": "192.168.0.69",
            "StaticIPv4SubNetMask": "255.255.255.0",
            "StaticIPv4Gateway": "192.168.0.201",
            "UseDHCP": "false"
        },
        "Apps": [
            {
[... 79 lines cut ...]
                    "FrameRate": "5",
                    "MinimumAmplitude": "42",
                    "Resolution": "0",
                    "ClippingCuboid": "{\"XMin\": -3.402823e+38, \"XMax\": 3.402823e+38, \"YMin\": -3.402823e+38, \"YMax\": 3.402823e+38, \"ZMin\": -3.402823e+38, \"ZMax\": 3.402823e+38}",
                    "SpatialFilterType": "0",
                    "SymmetryThreshold": "0",
                    "TemporalFilterType": "0",
                    "ThreeFreqMax2FLineDistPercentage": "80",
                    "ThreeFreqMax3FLineDistPercentage": "80",
                    "TwoFreqMaxLineDistPercentage": "80",
                    "Type": "upto30m_moderate",
                    "MaxAllowedLEDFrameRate": "11.9",
                    "SpatialFilter": {
                        "Type": "0",
                        "TypeStr_": "Off"
                    },
                    "TemporalFilter": {
                        "Type": "0",
                        "TypeStr_": "Off"
                    }
                }
            }
        ]
    }
}

tpanzarella, 2017-12-21:
Perhaps you need to normalize/scale the dynamic range of the values so they render nicely as a PNG for visualization purposes.

niallomahony93, 2017-12-21:
Ah that sorted it! Thank you for your prompt reply.

closed 2017-12-21 by niallomahony93