ifm/libo3d3xx #116: issue with building the oem partition

mikeNike9, 2017-12-21:
Here is the out put that has the error.
from first glance seems that 

CMake Error at CMakeLists.txt:68 (find_package):
  By not providing "Findresultsync.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "resultsync", but CMake did not find one.

  Could not find a package configuration file provided by "resultsync"
  (requested version 0.2) with any of the following names:

    resultsyncConfig.cmake
    resultsync-config.cmake

  Add the installation prefix of "resultsync" to CMAKE_PREFIX_PATH or set
  "resultsync_DIR" to a directory containing one of the above files.  If
  "resultsync" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/mike/libo3d3xx/modules/oem/build/CMakeFiles/CMakeOutput.log".
See also "/home/mike/libo3d3xx/modules/oem/build/CMakeFiles/CMakeError.log".

find_package(resultsync 0.2 REQUIRED) line 68

Where do I get this package.
Or do I even need it.

mikeNike9, 2017-12-21:
I cannot seem to find Lib_resultsync anywhere in my filesystem.
Was this something that should have been sourced when I initially build the environment because I am using an older version of  poky-glibc-i686-vantage-image-armv7ahf-vfp-neon-toolchain-qte-1.8.1.sh.
This script was given to me along with the gem tool months ago though.

graugans, 2017-12-29:
@mikeNike9 there is no need to rebuild the resultsync library by zour own this is already included in the latest firmware release and within the matching toolchain. So it is maybe needed to update the firmware plus the toolchain. Sorry I can not provide any of them because I am out of office until January the 8th. But zou can contact Ville who will be in the office starting with the 2nd of January.

closed 2020-07-27 by theseankelly