ifm/libo3d3xx #121: Cross compile error with xmlrpc

zschira, 2018-02-14:
I seem to be having a linking issue with libxmlrpc when attempting to cross compile for a raspberry pi. I have tried installing libxmlrpc on my pi from source and from pre-built binaries, but I am getting the same issue. Each time I try to compile the camera module I get the following error: 
```
/home/zach/rpi/root/usr/local/lib/libxmlrpc_client++.so: undefined reference to `xmlrpc_c::packetSocket::writeWait(xmlrpc_c::packetPtr const&, bool*) const'
/home/zach/rpi/root/usr/local/lib/libxmlrpc++.so: undefined reference to `xmlrpc_string_validate'
collect2: error: ld returned 1 exit status
src/bin/CMakeFiles/o3d3xx-version.dir/build.make:100: recipe for target 'src/bin/o3d3xx-version' failed
make[2]: *** [src/bin/o3d3xx-version] Error 1
CMakeFiles/Makefile2:289: recipe for target 'src/bin/CMakeFiles/o3d3xx-version.dir/all' failed
make[1]: *** [src/bin/CMakeFiles/o3d3xx-version.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
```
My host pc is running Ubuntu 17.10, and my toolchain file is as follows
```
set(CMAKE_SYSTEM_NAME Linux)
set(CMAKE_SYSROOT /home/zach/rpi/root)

set(CROSSTOOL_ROOT /home/zach/rpi/gcc-linaro-4.9-2016.02-x86_64_arm-linux-gnueabihf)
set(CMAKE_C_COMPILER "${CROSSTOOL_ROOT}/bin/arm-linux-gnueabihf-gcc")
set(CMAKE_CXX_COMPILER "${CROSSTOOL_ROOT}/bin/arm-linux-gnueabihf-g++")

set(CMAKE_FIND_ROOT_PATH ${CMAKE_SYSROOT})
set(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_PACKAGE ONLY)

set(CROSSTOOL_EXE_LINKER_FLAGS
    "-Wl,-rpath,${CMAKE_SYSROOT}/lib:${CMAKE_SYSROOT}/lib/arm-linux-gnueabihf")

set(CPACK_DEBIAN_PACKAGE_ARCHITECTURE "armhf")
```

I've tried explicitly specifying flags to link xmlrpc, but everything I do seems to produce the same results. Thanks for any help

graugans, 2018-02-14:
@zschira I have no out of the box solution for you. There a few things you could try:

- Doing try a native build on the PI itself it may take quite a long time. There are make targets for creating debian packages.
-  You can check the resulting library with ``nm`` or ``objdump``  if it contains the symbols you do need.
- Are you sure you have crosscompiled the ``xmlrpc-c`` library?
- We do provide a [xmlrpc-c library with CMake](https://github.com/graugans/xmlrpc-c) support for the Windows port. So you can give it a try if it will build wit all symbols included. I have not tried to use it on Linux latley.
In anycase it would be really nice to get your solution documented here so others can participate.

zschira, 2018-02-15:
Thanks for the response. I was successfully able to do a native build on the Pi. I might keep trying to find a solution for cross compiling, but for now this works.

closed 2020-07-27 by theseankelly