ifm/libo3d3xx #124: libo3d3xx build error

mikeNike9, 2018-03-14:
I am trying to build the newest version of libo3d3xx.
I have everything setup on my computer to work well with ifm3D and older version of libo3d3xx.
I try to run make and get these set of errors when trying to build the o3d3xx-viewer
is this common and should I disable the viewer from being built.

[ 28%] Automatic moc for target o3d3xx-viewer
[ 28%] Built target o3d3xx-viewer_automoc
[ 35%] Linking CXX executable o3d3xx-viewer
//usr/lib/x86_64-linux-gnu/libxmlrpc_client.so.3: undefined reference to `curl_slist_free_all@CURL_OPENSSL_3'
//usr/lib/x86_64-linux-gnu/libxmlrpc_client.so.3: undefined reference to `curl_easy_setopt@CURL_OPENSSL_3'
//usr/lib/x86_64-linux-gnu/libxmlrpc_client.so.3: undefined reference to `curl_multi_cleanup@CURL_OPENSSL_3'
//usr/lib/x86_64-linux-gnu/libxmlrpc_client.so.3: undefined reference to `curl_easy_cleanup@CURL_OPENSSL_3'
//usr/lib/x86_64-linux-gnu/libxmlrpc_client.so.3: undefined reference to `curl_slist_append@CURL_OPENSSL_3'
//usr/lib/x86_64-linux-gnu/libxmlrpc_client.so.3: undefined reference to `curl_multi_add_handle@CURL_OPENSSL_3'
//usr/lib/x86_64-linux-gnu/libxmlrpc_client.so.3: undefined reference to `curl_easy_init@CURL_OPENSSL_3'
//usr/lib/x86_64-linux-gnu/libxmlrpc_client.so.3: undefined reference to `curl_easy_strerror@CURL_OPENSSL_3'
//usr/lib/x86_64-linux-gnu/libxmlrpc_client.so.3: undefined reference to `curl_multi_init@CURL_OPENSSL_3'
//usr/lib/x86_64-linux-gnu/libxmlrpc_client.so.3: undefined reference to `curl_multi_strerror@CURL_OPENSSL_3'
//usr/lib/x86_64-linux-gnu/libxmlrpc_client.so.3: undefined reference to `curl_global_cleanup@CURL_OPENSSL_3'
//usr/lib/x86_64-linux-gnu/libxmlrpc_client.so.3: undefined reference to `curl_easy_getinfo@CURL_OPENSSL_3'
/home/mike/anaconda3/lib/libpng16.so.16: undefined reference to `inflateValidate@ZLIB_1.2.9'
//usr/lib/x86_64-linux-gnu/libxmlrpc_client.so.3: undefined reference to `curl_multi_remove_handle@CURL_OPENSSL_3'
//usr/lib/x86_64-linux-gnu/libxmlrpc_client.so.3: undefined reference to `curl_global_init@CURL_OPENSSL_3'
//usr/lib/x86_64-linux-gnu/libxmlrpc_client.so.3: undefined reference to `curl_multi_info_read@CURL_OPENSSL_3'
//usr/lib/x86_64-linux-gnu/libxmlrpc_client.so.3: undefined reference to `curl_multi_perform@CURL_OPENSSL_3'
//usr/lib/x86_64-linux-gnu/libxmlrpc_client.so.3: undefined reference to `curl_version_info@CURL_OPENSSL_3'
//usr/lib/x86_64-linux-gnu/libxmlrpc_client.so.3: undefined reference to `curl_multi_fdset@CURL_OPENSSL_3'
collect2: error: ld returned 1 exit status
src/bin/CMakeFiles/o3d3xx-viewer.dir/build.make:607: recipe for target 'src/bin/o3d3xx-viewer' failed
make[5]: *** [src/bin/o3d3xx-viewer] Error 1
CMakeFiles/Makefile2:241: recipe for target 'src/bin/CMakeFiles/o3d3xx-viewer.dir/all' failed
make[4]: *** [src/bin/CMakeFiles/o3d3xx-viewer.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMakeFiles/O3D3XX_IMAGE.dir/build.make:116: recipe for target 'O3D3XX_IMAGE-prefix/src/O3D3XX_IMAGE-stamp/O3D3XX_IMAGE-build' failed
make[2]: *** [O3D3XX_IMAGE-prefix/src/O3D3XX_IMAGE-stamp/O3D3XX_IMAGE-build] Error 2
CMakeFiles/Makefile2:179: recipe for target 'CMakeFiles/O3D3XX_IMAGE.dir/all' failed
make[1]: *** [CMakeFiles/O3D3XX_IMAGE.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

graugans, 2018-03-15:
Hello @mikeNike9 what operating system are you using? I have created a [Dockerfile](https://github.com/graugans/docker_images/blob/master/libo3d3xx/Dockerfile) for Ubuntu 16.04. It looks like you are mixing up system libraries with libraries provided by anaconda. I had a similar problem with my installation. This was why I've created the Docker container to avoid such conflicts.

CommonGibbon, 2018-07-25:
To resolve the "/home/mike/anaconda3/lib/libpng16.so.16: undefined reference to inflateValidate@ZLIB_1.2.9' " error try the following:
Prior to running 'make' temporarily rename your ~/anaconda3 folder to something else, such as "anaconda3_temp". Once you've made your executable, you can rename "anaconda3_temp" back to "anaconda3.

closed 2020-07-27 by theseankelly