ifm/libo3d3xx #127: make error: 6 tests failed from FrameGrabberTest

sharathrjtr, 2018-08-03:
Hi, 
I've Ubuntu 16.04 with g++ (Ubuntu 5.4.0-6ubuntu1~16.04.10), pcl version 1.7, opencv version 3.0 and the sensor is installed with Firmwareware ver 1.6.2114.
I get the following error while trying to build the package. I've tried on all the libo3d3xx versions from 0.4.9 to 0.7.3 and all of them gave the same error.

```
[ RUN      ] FrameGrabberTest.SoftwareTrigger
[       OK ] FrameGrabberTest.SoftwareTrigger (5502 ms)
[ RUN      ] FrameGrabberTest.SoftwareTriggerMultipleClients
[       OK ] FrameGrabberTest.SoftwareTriggerMultipleClients (7542 ms)
[ RUN      ] FrameGrabberTest.SoftwareTriggerInFreeRunMode
[       OK ] FrameGrabberTest.SoftwareTriggerInFreeRunMode (5478 ms)
[----------] 9 tests from FrameGrabberTest (67766 ms total)

[----------] Global test environment tear-down
[==========] 9 tests from 1 test case ran. (67766 ms total)
[  PASSED  ] 3 tests.
[  FAILED  ] 6 tests, listed below:
[  FAILED  ] FrameGrabberTest.WaitForFrame
[  FAILED  ] FrameGrabberTest.WaitForFrameFast
[  FAILED  ] FrameGrabberTest.CustomSchemas
[  FAILED  ] FrameGrabberTest.ByteBufferCopyCtor
[  FAILED  ] FrameGrabberTest.ByteBufferCopyAssignmentOperator
[  FAILED  ] FrameGrabberTest.FrameGrabberRecycling

 6 FAILED TESTS
test/CMakeFiles/check.dir/build.make:57: recipe for target 'test/CMakeFiles/check' failed
make[6]: *** [test/CMakeFiles/check] Error 1
CMakeFiles/Makefile2:369: recipe for target 'test/CMakeFiles/check.dir/all' failed
make[5]: *** [test/CMakeFiles/check.dir/all] Error 2
CMakeFiles/Makefile2:376: recipe for target 'test/CMakeFiles/check.dir/rule' failed
make[4]: *** [test/CMakeFiles/check.dir/rule] Error 2
Makefile:262: recipe for target 'check' failed
make[3]: *** [check] Error 2
CMakeFiles/O3D3XX_FRAMEGRABBER.dir/build.make:120: recipe for target 'O3D3XX_FRAMEGRABBER-prefix/src/O3D3XX_FRAMEGRABBER-stamp/O3D3XX_FRAMEGRABBER-test' failed
make[2]: *** [O3D3XX_FRAMEGRABBER-prefix/src/O3D3XX_FRAMEGRABBER-stamp/O3D3XX_FRAMEGRABBER-test] Error 2
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/O3D3XX_FRAMEGRABBER.dir/all' failed
make[1]: *** [CMakeFiles/O3D3XX_FRAMEGRABBER.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

```
Could you please help me.

graugans, 2018-08-03:
@sharathrjtr may I recommend to use the [ifm3d library](https://github.com/lovepark/libo3d3xx/issues/127) which is the successor of the libo3d3xx and recommended for new projects. From my experience most of the issues during the the ``make test`` run is caused by a network issue. For the ifm3d library we do have a [trouble shooting guide](https://github.com/lovepark/ifm3d/blob/dev-0.9.3/doc/troubleshoot.md) in the current development branch which describes how to solve those issues. The basic test is, can you ping the camera? In case you are not sure what IP address the camera does have you can figure out the IP address.  Eithe with the Windows GUI or from the linux command line.

```
 $ avahi-browse -arp 2>/dev/null | grep  ^=.*O3D.*Workstation 
```

sharathrjtr, 2018-08-03:
@graugans thanks for quick reply. The IP address was set to the default IP 192.168.0.69, got it from the IFM vision assistant.

I downloaded the ifm3d library and installed it properly using the [alternative build procedure](https://github.com/lovepark/ifm3d#the-default-build).

There was only one hiccup while installing ifm3d, thought you should know.
For the first time when i ran `make check` command. I got the error
```
[----------] Global test environment tear-down
[==========] 10 tests from 1 test case ran. (12534 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 10 tests, listed below:
[  FAILED  ] Image.MoveCtor
[  FAILED  ] Image.MoveAssignmentOperator
[  FAILED  ] Image.CopyCtor
[  FAILED  ] Image.CopyAssignmentOperator
[  FAILED  ] Image.References
[  FAILED  ] Image.CloudMechanics
[  FAILED  ] Image.XYZImage
[  FAILED  ] Image.ComputeCartesian
[  FAILED  ] Image.TimeStamp
[  FAILED  ] Image.IlluTemp

10 FAILED TESTS
modules/image/test/CMakeFiles/check_image.dir/build.make:57: recipe for target 'modules/image/test/CMakeFiles/check_image' failed
make[3]: *** [modules/image/test/CMakeFiles/check_image] Error 1
CMakeFiles/Makefile2:741: recipe for target 'modules/image/test/CMakeFiles/check_image.dir/all' failed
make[2]: *** [modules/image/test/CMakeFiles/check_image.dir/all] Error 2
CMakeFiles/Makefile2:76: recipe for target 'CMakeFiles/check.dir/rule' failed
make[1]: *** [CMakeFiles/check.dir/rule] Error 2
Makefile:184: recipe for target 'check' failed
make: *** [check] Error 2
```

I felt to check individually in the following order with all tests getting passed.
```
make -j8 check_camera
make -j8 check_fg
make -j8 check_image
```
And later i ran `make check` without any errors and all tests getting passed. This might be of use to someone else.

Thanks for your help :+1:

closed 2018-08-03 by sharathrjtr