ifm/libo3d3xx #130: Test fail with 1.23.1522 on O3D301

s-hemer, 2018-08-20:
After installling the latest firmware, I wanted to build 0.7.5 on 16.04.5 but I get 5 failed test:

o3d3xx-camera-tests.cpp:182: Failure
Value of: 35
Expected: params.size()
Which is: 38
[  FAILED  ] Camera_Tests.GetDeviceConfig (566 ms)

o3d3xx-app-imager-tests.cpp:131: Failure
Value of: 9
Expected: params.size()
Which is: 10
[  FAILED  ] AppImagerTest.GetAppParameters (3599 ms)

o3d3xx-app-imager-tests.cpp:306: Failure
Expected: params.at("ExposureTime") doesn't throw an exception.
  Actual: it throws.
[  FAILED  ] AppImagerTest.GetImagerParameters (3887 ms)

3d3xx-app-imager-tests.cpp:425: Failure
Expected: cam_->SetImagerConfig(im.get()) doesn't throw an exception.
  Actual: it throws.
[  FAILED  ] AppImagerTest.ImagerConfig (4477 ms)

o3d3xx-app-imager-tests.cpp:590: Failure
Expected: cam_->SetImagerConfig(im.get()) doesn't throw an exception.
  Actual: it throws.
[  FAILED  ] AppImagerTest.Exposure (3946 ms)

Unfortunately, the ifm3d lib also fails with

ifm3d-image-tests.cpp:421: Failure
Value of: std::equal(x_cam.begin<std::int16_t>(), x_cam.end<std::int16_t>(), x_computed.begin<std::int16_t>(), cmp)
  Actual: false
Expected: true
[  FAILED  ] Image.ComputeCartesian (217 ms)

which stops me from trying to switch over (I pretty much like the feature to build up a Custom Image Container).

Reset several time with the latest ifmVisionAssistant (1.8.23).

graugans, 2018-08-20:
@s-hemer the failed ifm3d test can be caused by invalid pixels. Can you please check this? The libo3d3xx I will check tomorrow when I have a camera available

s-hemer, 2018-08-21:
For the ifm3d error, it was indeed because of invalid pixels. Thanks for the hint, never checked before if there are invalid pixels present when doing the build checks.

The libo3d3xx error yet persist also without invalid pixel.

cveezee, 2018-08-31:
@s-hemer the libo3d3xx errors are due to the fact that the tests haven't been updated for the version of firmware you are using .

if possible, it is highly recommended that you use the **ifm3d** instead of the libo3d3xx

closed 2020-07-27 by theseankelly