ifm/libo3d3xx #131: Installation failing on Ubuntu 18.0.4

inbangsa, 2018-09-14:
I am trying to build libo3d3xx on ubuntu 18.0.4 from the base directory Camera module build successfully  but issue occurred during installation step. 

following is the error log
```
(Reading database ... 254568 files and directories currently installed.)
Preparing to unpack libo3d3xx-camera_0.7.5_ubuntu-18.04.1_amd64.deb ...
Unpacking libo3d3xx-camera (0.7.5) ...
dpkg: error processing archive libo3d3xx-camera_0.7.5_ubuntu-18.04.1_amd64.deb (--install):
 trying to overwrite '/usr/include/gtest/gtest-death-test.h', which is also in package googletest:amd64 1.8.0-6
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Errors were encountered while processing:
 libo3d3xx-camera_0.7.5_ubuntu-18.04.1_amd64.deb
CMakeFiles/O3D3XX_CAMERA.dir/build.make:77: recipe for target 'O3D3XX_CAMERA-prefix/src/O3D3XX_CAMERA-stamp/O3D3XX_CAMERA-install' failed
make[2]: *** [O3D3XX_CAMERA-prefix/src/O3D3XX_CAMERA-stamp/O3D3XX_CAMERA-install] Error 1
CMakeFiles/Makefile2:142: recipe for target 'CMakeFiles/O3D3XX_CAMERA.dir/all' failed
make[1]: *** [CMakeFiles/O3D3XX_CAMERA.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
 ```

This needs the fix in .deb package

jorgenfb, 2018-10-16:
I have the same experience (ubuntu 18.04):
```
dpkg: error processing archive libo3d3xx-camera_0.7.0_ubuntu-18.04.1_amd64.deb (--install):
 trying to overwrite '/usr/include/gtest/gtest-death-test.h', which is also in package googletest:amd64 1.8.0-6
```

jorgenfb, 2018-10-16:
I found workaround. Just disable building the tests, then it won't be included anymore:
```
cmake -DBUILD_TESTS=OFF
```

Another tips for ubuntu 18.04 when building the image module is to not build the viewer if not needed. It depends on libpcl1.7 which isn't available by default.
```
cmake -DBUILD_EXE_VIEWER=OFF -DBUILD_TESTS=OFF
```

graugans, 2018-10-16:
@jorgenfb there is a Pull-Request #128 available which is currently can not be merged. Maybe @inbangsa can take of the merge conflict.

closed 2020-07-27 by theseankelly