ifm/libo3d3xx #136: Error in make

matteofiorini, 2019-07-23:
Hi,
I get an error in the make of libo3d3xx as I attached in the txt file.
I have ubuntu 16.04 LTS and Ros Kinetic installed. I have plugged in the ethernet port a O3D302 which I already tested in windows. 
The IP of the camera is 169.254.89.234
So i set the IP of my PC to the same with the subnet mask 255.255.255.0
The command ping 169.254.89.234 returns after a while this:
--- 169.254.89.234 ping statistics ---
9 packets transmitted, 9 received, 0% packet loss, time 8182ms
rtt min/avg/max/mdev = 0.031/0.035/0.040/0.007 ms
So the connection seems to work. 
Should I change something in the src files in order to pass this IP address?

matteofiorini, 2019-07-23:
This is the link of the txt file, I don't know why it does not upload it.
https://file.io/TpoAMR

matteofiorini, 2019-07-23:
This is the link for the txt file, I don't know why it does not upload it.
https://pastebin.com/YNpSJwRm

graugans, 2019-07-23:
Looks like ``LIB_glog`` is missing. Have you installed Googles logging library? Is there any specific reason why you have chosen libo3d3xx and not the [ifm3d](https://github.com/ifm/ifm3d)?

tpanzarella, 2019-07-23:
@matteofiorini To follow up on what @graugans says, I would highly suggest you use [ifm3d](https://github.com/ifm/ifm3d) instead of this project. `ifm3d` is the successor of this project and will soon deprecate it. That said, whether you use this project or `ifm3d` you will still need Google Glog installed as it is the logging framework used by both projects. If you do choose to move to `ifm3d`, you can avoid the compilation step completely by using our pre-built binaries (looks like you are on Ubuntu 16.04, which we supply binaries for) -- instructions are at the [ifm3d README](https://github.com/ifm/ifm3d/blob/master/README.md). Of course, if you prefer, you can [build from source](https://github.com/ifm/ifm3d/blob/master/doc/source_build.md).

matteofiorini, 2019-07-23:
Thank you both, I will move to ifm3d as you suggest.

graugans, 2019-09-11:
@matteofiorini can we close this issue?

closed 2020-07-27 by theseankelly