ifm/libo3d3xx #137: Error in image module make

PatronusCoder, 2019-12-04:
I am getting the following error while trying to build the image module for this library. I have attached a text file showing the outputs of cmake and make also. 

[image_module_error.txt](https://github.com/ifm/libo3d3xx/files/3921211/image_module_error.txt)

```console
Scanning dependencies of target o3d3xx_image
[  8%] Building CXX object src/libo3d3xx_image/CMakeFiles/o3d3xx_image.dir/image.cpp.o
[  8%] Built target o3d3xx_image
Scanning dependencies of target o3d3xx_image_static
[ 16%] Linking CXX static library libo3d3xx_image_static.a
[ 16%] Built target o3d3xx_image_static
Scanning dependencies of target o3d3xx_image_shared
[ 25%] Linking CXX shared library libo3d3xx_image.so
[ 25%] Built target o3d3xx_image_shared
Scanning dependencies of target o3d3xx-viewer
[ 33%] Building CXX object src/bin/CMakeFiles/o3d3xx-viewer.dir/o3d3xx-viewer.cpp.o
[ 41%] Linking CXX executable o3d3xx-viewer
/usr/bin/ld: CMakeFiles/o3d3xx-viewer.dir/o3d3xx-viewer.cpp.o: undefined reference to symbol '_ZN5boost6system15system_categoryEv'
//usr/lib/x86_64-linux-gnu/libboost_system.so.1.58.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
src/bin/CMakeFiles/o3d3xx-viewer.dir/build.make:122: recipe for target 'src/bin/o3d3xx-viewer' failed
make[2]: *** [src/bin/o3d3xx-viewer] Error 1
CMakeFiles/Makefile2:239: recipe for target 'src/bin/CMakeFiles/o3d3xx-viewer.dir/all' failed
make[1]: *** [src/bin/CMakeFiles/o3d3xx-viewer.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

```
I can't understand the error in the first place to be able to resolve it. It would be great if someone could help.

theseankelly, 2019-12-04:
Hi @PatronusCoder -- first off, I suggest you use the [`ifm3d`](https://github.com/ifm/ifm3d) library instead. `libo3d3xx` is an older implementation, kept for legacy reasons. Newer developments should be done on `ifm3d` -- we also provide binaries of `ifm3d` for Ubuntu through apt.

Does this work for you or do you have specific need for `libo3d3xx`? If so, can you provide some more details about which OS and compiler you're trying to compile with?

PatronusCoder, 2019-12-05:
Hey @theseankelly .. My whole code was based upon modules from the `libo3d3xx` library. How different is the `ifm3d` library from it?  Because if the difference is signiificant from the previous implementation, I would prefer to resolve issue with running `libo3d3xx` library. I am currently using Ubuntu 16.04 on Parallels for Mac virtual machine with following compilers:
```console
ii  g++                                        4:5.3.1-1ubuntu1                                amd64        GNU C++ compiler
ii  g++-5                                      5.4.0-6ubuntu1~16.04.12                         amd64        GNU C++ compiler
ii  gcc                                        4:5.3.1-1ubuntu1                                amd64        GNU C compiler
ii  gcc-5                                      5.4.0-6ubuntu1~16.04.12                         amd64        GNU C compiler
ii  hardening-includes                         2.7ubuntu2                                      all          Makefile for enabling compiler flags for security hardening
ii  libllvm3.8:amd64                           1:3.8-2ubuntu4                                  amd64        Modular compiler and toolchain technologies, runtime library
ii  libllvm6.0:amd64                           1:6.0-1ubuntu2~16.04.1                          amd64        Modular compiler and toolchain technologies, runtime library
ii  libxkbcommon0:amd64                        0.5.0-1ubuntu2.1                                amd64        library interface to the XKB compiler - shared library

```

theseankelly, 2019-12-12:
Hi @PatronusCoder,

The `ifm3d` library is intended to be nearly identical from an API perspective; it ought to be a drop-in replacement for your implementation.

That said, based on the error log it looks like Boost::System is not being linked properly. I've pushed a potential fix to [this branch](https://github.com/theseankelly/libo3d3xx/tree/boost_link). Unfortunately I cannot reproduce your issue on a clean install of Ubuntu 16.04 so I can't prove this is a fix for your issues, but based on the link errors it seems likely.

Let me know if it works and I'll merge the change in to master ASAP!

PatronusCoder, 2019-12-16:
Hi @theseankelly ,

This fix did not solve my issue. I got the following error:

```console
Scanning dependencies of target o3d3xx_image
[ 20%] Building CXX object src/libo3d3xx_image/CMakeFiles/o3d3xx_image.dir/image.cpp.o
[ 20%] Built target o3d3xx_image
Scanning dependencies of target o3d3xx_image_static
[ 40%] Linking CXX static library libo3d3xx_image_static.a
[ 40%] Built target o3d3xx_image_static
Scanning dependencies of target o3d3xx_image_shared
[ 60%] Linking CXX shared library libo3d3xx_image.so
[ 60%] Built target o3d3xx_image_shared
Scanning dependencies of target o3d3xx-viewer
[ 80%] Building CXX object src/bin/CMakeFiles/o3d3xx-viewer.dir/o3d3xx-viewer.cpp.o
[100%] Linking CXX executable o3d3xx-viewer
/usr/bin/ld: CMakeFiles/o3d3xx-viewer.dir/o3d3xx-viewer.cpp.o: undefined reference to symbol '_Z37vtkRenderingOpenGL2_AutoInit_Destructv'
/usr/local/lib/libvtkRenderingOpenGL2-8.2.so.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
src/bin/CMakeFiles/o3d3xx-viewer.dir/build.make:114: recipe for target 'src/bin/o3d3xx-viewer' failed
make[2]: *** [src/bin/o3d3xx-viewer] Error 1
CMakeFiles/Makefile2:217: recipe for target 'src/bin/CMakeFiles/o3d3xx-viewer.dir/all' failed
make[1]: *** [src/bin/CMakeFiles/o3d3xx-viewer.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
```

theseankelly, 2019-12-16:
Hi @PatronusCoder,

Sorry for the back and forth, but it appears your environment and/or installed libraries have deviated from the standard Ubuntu 16.04 installation. This isn't a problem, but it's also challenging to debug remotely.

Here's another option: all of your link errors have been in the o3d3xx-viewer application. This is a GUI for visualizing point clouds and is not necessary for developing applications based on the library. You could also try skipping this application by specifying `-DBUILD_EXE_VIEWER=OFF` when compiling the `image` module.

closed 2020-06-03 by PatronusCoder