ifm/libo3d3xx #139: can you offer 16.04 *.deb

comwise, 2020-05-21:
i tried build source code, but occur some problem,  can you offer 16.04 *.deb directly? 
thanks for your reply

theseankelly, 2020-05-21:
Hi @comwise -- I recommend you use our newer library, [ifm3d](https://github.com/ifm/ifm3d). The README `ifm3d` has instructions for setting up an apt repository for Ubuntu 16.04.

comwise, 2020-05-22:
> Hi @comwise -- I recommend you use our newer library, [ifm3d](https://github.com/ifm/ifm3d). The README for `ifm3d` has instructions for setting up an apt repository for Ubuntu 16.04.

the ifm3d lib support o3d3xx-ros?  i build this ros project which need libo3d3xx lib?

theseankelly, 2020-05-22:
libo3d3xx and o3d3xx-ros go together, ifm3d and ifm3d-ros go together. So yes `ifm3d-ros` is built on top of `ifm3d` and supports the O3D3XX and O3X cameras.

By the way, if you're using Kinetic or Melodic, we're in the ROS distribution. You can do `apt-get install ros-melodic-ifm3d` or `apt-get install ros-kinetic-ifm3d` to install the ROS driver and core `ifm3d` library.

comwise, 2020-05-27:
@theseankelly thanks for your replay , ifm3d-ros is good for o3d3xx

theseankelly, 2020-05-27:
Great! feel free to file issues over on those projects if you run into issues.

closed 2020-05-27 by theseankelly