ifm/libo3d3xx #140: Camera communication (high ping packets loss)

robinbansal08, 2020-06-03:
I'm running Ubuntu 16.04 together with the O3D303 camera. 

A ping to the camera IP 192.168.0.51 shows 90% packet loss:

```
$ ping 192.168.0.51
PING 192.168.0.51 (192.168.0.51) 56(84) bytes of data.
64 bytes from 192.168.0.51: icmp_seq=26 ttl=64 time=0.164 ms
64 bytes from 192.168.0.51: icmp_seq=27 ttl=64 time=0.214 ms
64 bytes from 192.168.0.51: icmp_seq=28 ttl=64 time=0.218 ms
64 bytes from 192.168.0.51: icmp_seq=68 ttl=64 time=0.280 ms
64 bytes from 192.168.0.51: icmp_seq=69 ttl=64 time=0.163 ms
64 bytes from 192.168.0.51: icmp_seq=70 ttl=64 time=0.262 ms
64 bytes from 192.168.0.51: icmp_seq=71 ttl=64 time=0.169 ms
64 bytes from 192.168.0.51: icmp_seq=72 ttl=64 time=0.192 ms
64 bytes from 192.168.0.51: icmp_seq=73 ttl=64 time=0.179 ms
64 bytes from 192.168.0.51: icmp_seq=74 ttl=64 time=0.196 ms
64 bytes from 192.168.0.51: icmp_seq=75 ttl=64 time=0.138 ms
64 bytes from 192.168.0.51: icmp_seq=76 ttl=64 time=0.172 ms
64 bytes from 192.168.0.51: icmp_seq=77 ttl=64 time=0.141 ms
64 bytes from 192.168.0.51: icmp_seq=115 ttl=64 time=0.399 ms
64 bytes from 192.168.0.51: icmp_seq=116 ttl=64 time=0.168 ms
64 bytes from 192.168.0.51: icmp_seq=121 ttl=64 time=0.162 ms
64 bytes from 192.168.0.51: icmp_seq=155 ttl=64 time=0.324 ms
64 bytes from 192.168.0.51: icmp_seq=184 ttl=64 time=0.261 ms
64 bytes from 192.168.0.51: icmp_seq=189 ttl=64 time=0.149 ms
^C
--- 192.168.0.51 ping statistics ---
196 packets transmitted, 19 received, 90% packet loss, time 199685ms
rtt min/avg/max/mdev = 0.138/0.207/0.399/0.069 ms

```

When trying to use libo3d3xx command line utilities commands, most of the times, it fails with XMRPC call timed out error.
```
pactrispro@CLOUDLink:~/xpsSea$ o3d3xx-ls --ip 192.168.0.51
Failed to list applications: XMLRPC call timed out
pactrispro@CLOUDLink:~/xpsSea$ o3d3xx-ls --ip 192.168.0.51
Failed to list applications: XMLRPC call timed out
pactrispro@CLOUDLink:~/xpsSea$ o3d3xx-ls --ip 192.168.0.51
* [1] id=1186278907, name=Box Detection Application, description=
pactrispro@CLOUDLink:~/xpsSea$ o3d3xx-ls --ip 192.168.0.51
Failed to list applications: XMLRPC call timed out
pactrispro@CLOUDLink:~/xpsSea$ o3d3xx-ls --ip 192.168.0.51
Failed to list applications: XMLRPC call timed out
pactrispro@CLOUDLink:~/xpsSea$ o3d3xx-ls --ip 192.168.0.51
Failed to list applications: XMLRPC call timed out
pactrispro@CLOUDLink:~/xpsSea$ o3d3xx-ls --ip 192.168.0.51
Failed to list applications: XMLRPC call timed out

```

I have already tried with rebooting, resetting the camera, but problem still persists. 

There are several other O3D303 cameras connected in the same network, but they are working fine.

`
pactrispro@CLOUDLink:~/xpsSea$ o3d3xx-dump --ip 192.168.0.51 > camera_51_dump.json`

Here is the result of it:

```

{
    "o3d3xx": {
        "libo3d3xx": "705",
        "Date": "Wed Jun  3 00:30:54 2020",
        "HWInfo": {
            "Connector": "#!02_A300_C40_04246016_008023176",
            "Diagnose": "#!02_D322_C34_04068012_008026824",
[... 98 lines cut ...]
                    "ClippingCuboid": "{\"XMin\": -3.402823e+38, \"XMax\": 3.402823e+38, \"YMin\": -3.402823e+38, \"YMax\": 3.402823e+38, \"ZMin\": -3.402823e+38, \"ZMax\": 3.402823e+38}",
                    "SpatialFilterType": "1",
                    "SymmetryThreshold": "0",
                    "TemporalFilterType": "0",
                    "ThreeFreqMax2FLineDistPercentage": "80",
                    "ThreeFreqMax3FLineDistPercentage": "80",
                    "TwoFreqMaxLineDistPercentage": "80",
                    "Type": "upto30m_moderate",
                    "MaxAllowedLEDFrameRate": "4.9000000000000004",
                    "SpatialFilter": {
                        "Type": "1",
                        "TypeStr_": "Median Filter",
                        "MaskSize": "1",
                        "MaskSizeStr_": "5x5"
                    },
                    "TemporalFilter": {
                        "Type": "0",
                        "TypeStr_": "Off"
                    }
                }
            }
        ]
    }
}
```

theseankelly, 2020-06-09:
Hey @robinbansal08 is it possible there's an IP address conflict on your network (perhaps DHCP Handed 192.168.0.55 to another device)? You could check by powering off the camera and seeing if ping works or fails.

Otherwise, the 1.6.2114 firmware release is several years old and it would be worth your time to upgrade and see if the issues persist.

@graugans any other thoughts from the embedded side?

graugans, 2020-06-09:
@robinbansal08 please check the following:

- Are there any O3D3xx devices with consecutive MAC addresses in your network. If so please remove all cameras which are close (+/-1) to the one you are testing. After this a full power cycle of any active network component is recommended.  A solution to this is a FW update to version 1.30.x
- Check if there is an issue with the Ethernet cable used. You can cross test with another working camera
- If there is any active network component, like a switch, in between please also power cycle them and try a different port.

If the error is still present please contact your local ifm sales responsible and initiate an RMA

robinbansal08, 2020-06-12:
Thank you for your replies.

@graugans You are correct regarding the MAC addresses issue (+/-1). This camera (192.168.0.51) with the high ping packets loss issue was working fine before, but this issue came up when I updated the firmware(to version 1.23.x) of few of the other cameras in the same network. Based on your suggestion, I have compared their MAC addresses and found one of the cameras (of which I updated the firmware) has consecutive sequence of the MAC address to the camera (192.168.0.51).

I have already replaced the camera(192.168.0.51) and now all the cameras in the network are working fine now.

I have also checked the MAC addresses of all the cameras in the same network and found around half of the cameras having MAC addresses +/- with other cameras. 

Currently all cameras are working fine but I want to update their firmware as we have to run our applications over the camera.  So if we update the firmware with the latest version 1.30.x, will this issue (ping packets loss) can still occur with the cameras with consecutive sequence MAC addresses ?

robinbansal08, 2020-06-12:
This issue is resolved, as per your suggestion I have updated the firmware of the cameras  to 1.30.5309 and the cameras with sequence of +/-1 MAC addresses are working fine.

graugans, 2020-06-15:
I am glad to hear the issue is solved. I'll close this, please reopen in case of any issues related to this topic.

closed 2020-06-15 by graugans