ifm/libo3d3xx #141: Random reappearance of noisy behaviour for the same scene

robinbansal08, 2020-06-12:
We have 48 O3D303 cameras installed in a hall of a warehouse at a height of 9.5 meters. 
All cameras are enabled as software triggered and are managed by a trigger scheduling algorithm to handle any interference issues.

We have noticed that sometimes all the 48 cameras at the same time captures very noisy point cloud of the same scene as compared to other times, when all the cameras provides good point cloud.

--> Here you see the nice point cloud captured by a camera. 
[image]

--> Here you see the noisy point cloud captured by the same camera for same scene the next day.
[image]

We did not change any filter configuration and there was no movement in the scene for both cases, still there much difference in the level of noise in both can be observed.

Here is the json file for the camera: [cam_ip_16_newswu_dump.txt](https://github.com/ifm/libo3d3xx/files/4771038/cam_ip_16_newswu_dump.txt)

GreNait, 2020-06-15:
Hi @robinbansal08,

First, could you tell us the firmware version of your units and the general mode? Like, do you use single/double exposure, do you use 1,2 or 3 frequencies? This might help us to verify the problem here.

The images itself are looking familiar to problems like sunlight or cross talk. If two or more cameras are active at the same time, and if their FOV is overlapping a bit, you get that kind of noise. 
Is the floor and ceiling reflective? We also had some issues, when the light is reflected by the floor to the ceiling and back (I don't expect that in this case, but you never know).

Best regards

graugans, 2020-06-15:
Hey @robinbansal08 unrelated to your issue, we do recommend to use https://github.com/ifm/ifm3d for new projects. The libo3d3xx project is not recommended for new projects unless you are planning to integrate code on the camera.

robinbansal08, 2020-06-15:
@graugans Thank you for your reply. Yes, we are using libo3d3xx as we want to integrate our codes on the camera.

@GreNait Thank you for your reply. Here is the dumped configuration file: 
```json
{
    "o3d3xx": {
        "libo3d3xx": "705",
        "Date": "Fri Jun 12 15:26:41 2020",
        "HWInfo": {
            "Connector": "#!02_A300_C40_04246016_008023176",
            "Diagnose": "#!02_D322_C34_04250651_008026824",
            "MiraSerial": "8670-81c3-0e63-0794",
            "MACAddress": "00:02:01:41:3B:AE",
            "Frontend": "#!02_F342_C34_19_00105_008023607",
            "Mainboard": "#!02_M381_C41_03901679_008023690",
            "Illumination": "#!02_I300_001_04201465_008001175"
        },
        "SWVersion": {
            "IFM_Software": "1.30.5309",
            "Main_Application": "1.0.44",
            "IFM_Recovery": "unversioned",
            "Algorithm_Version": "2.2.14",
            "Calibration_Device": "00:02:01:41:3b:ae",
            "Calibration_Version": "1.0.2",
            "Linux": "Linux version 3.14.34-rt31-yocto-standard-00048-gc8112a91aeb2-dirty (jenkins@dettlx190) (gcc version 4.9.2 (GCC) ) #1 SMP PREEMPT RT Wed Mar 27 18:25:10 CET 2019",
            "Diagnostic_Controller": "v1.0.44-2dcaabbc73-dirty"
        },
        "Device": {
            "Name": "New sensor",
            "Description": "",
            "ActiveApplication": "1",
            "PcicTcpPort": "50010",
            "PcicProtocolVersion": "3",
            "IOLogicType": "1",
            "IODebouncing": "true",
            "IOExternApplicationSwitch": "0",
            "SessionTimeout": "30",
            "ServiceReportPassedBuffer": "15",
            "ServiceReportFailedBuffer": "15",
            "ExtrinsicCalibTransX": "0",
            "ExtrinsicCalibTransY": "0",
            "ExtrinsicCalibTransZ": "0",
            "ExtrinsicCalibRotX": "0",
            "ExtrinsicCalibRotY": "0",
            "ExtrinsicCalibRotZ": "0",
            "EvaluationFinishedMinHoldTime": "10",
            "SaveRestoreStatsOnApplSwitch": "true",
            "IPAddressConfig": "0",
            "PasswordActivated": "false",
            "OperatingMode": "1",
            "DeviceType": "1:2",
            "ArticleNumber": "O3D303",
            "ArticleStatus": "AD",
            "UpTime": "3.8574999999999999",
            "ImageTimestampReference": "1569337495",
            "TemperatureFront1": "3276.6999999999998",
            "TemperatureFront2": "3276.6999999999998",
            "TemperatureIMX6": "33.009998321533203",
            "TemperatureIllu": "36.799999999999997",
            "PNIODeviceName": "",
            "EthernetFieldBus": "0",
            "EthernetFieldBusEndianness": "0",
            "EnableAcquisitionFinishedPCIC": "false"
        },
        "Net": {
            "MACAddress": "",
            "NetworkSpeed": "0",
            "StaticIPv4Address": "192.168.0.16",
            "StaticIPv4SubNetMask": "255.255.255.0",
            "StaticIPv4Gateway": "192.168.0.201",
[... 44 lines cut ...]
                    "ThreeFreqMax3FLineDistPercentage": "80",
                    "TwoFreqMaxLineDistPercentage": "80",
                    "Type": "upto30m_moderate",
                    "MaxAllowedLEDFrameRate": "4.9000000000000004",
                    "SpatialFilter": {
                        "Type": "1",
                        "TypeStr_": "Median Filter",
                        "MaskSize": "1",
                        "MaskSizeStr_": "5x5"
                    },
                    "TemporalFilter": {
                        "Type": "0",
                        "TypeStr_": "Off"
                    }
                }
            }
        ]
    }
}
```
The cameras are configured with `upto30m_moderate` mode. The cameras are installed at a height of 9.5 meters, inclined 10 degrees from the vertical.

There are no interference issues as we already have scheduled the camera software trigger using a central camera scheduling algorithm such that no two cameras (having common FOV) triggers at the same time.

The floor is reflective but not ceiling. This noisy behaviour is observed only sometimes during the day, but rest of the time the point clouds captured are without any noise. The warehouse where the cameras are installed is constructed such that there is no openings for direct entrance of the sunlight.

GreNait, 2020-06-17:
Okay, I have one other guess. Can you check your power supply and the cable length? We had some issues, that if the power supply is to low, but not low enough to be recognized by our system, that sometimes the image gets very noisy. 
This happens, if e.g. the camera is triggered and a peak is going through the powerline. 

Be aware, that you need at least 2A and 24V for every camera. With cable length more than 12m, you need to be certain that still the 24V are arriving at the camera.

Hope this helps.

closed 2020-06-30 by robinbansal08