ifm/libo3d3xx #142: Double free or corruption (out) error in framegrabber process

robinbansal08, 2020-07-25:
I am using libo3d3xx as our project requires deployment of the codes over the camera.

I am getting strange runtime errors **double free or corruption (out)** error  when using framegrabber function even with the simplest of the codes I tried as mentioned below. After various rounds of testing I found, that this error is quite inconsistent and disappears with commenting out segments of the code which are not even being used or called upon by any function calls (for example, function ```int getSavedPCD_(int id, pcl::PointCloud<pcl::PointXYZI>::Ptr& cloud_xyzi)``` in the code (Code: test_1) mentioned below). Interestingly, when I cross compiled the code to run it over the camera O3D303, it runs successfully over it without any runtime error. It would be really helpful if you could provide your expert insight about this strange error.

I have installed libo3d3xx 0.7.7 version and using Ubuntu 18.04. 

**My compiler list:**
```
ii  g++                                        4:7.4.0-1ubuntu2.3                               amd64        GNU C++ compiler
ii  g++-7                                      7.5.0-3ubuntu1~18.04                             amd64        GNU C++ compiler
ii  gcc                                        4:7.4.0-1ubuntu2.3                               amd64        GNU C compiler
ii  gcc-7                                      7.5.0-3ubuntu1~18.04                             amd64        GNU C compiler
ii  gfortran                                   4:7.4.0-1ubuntu2.3                               amd64        GNU Fortran 95 compiler
ii  gfortran-7                                 7.5.0-3ubuntu1~18.04                             amd64        GNU Fortran compiler
ii  libllvm6.0:amd64                           1:6.0-1ubuntu2                                   amd64        Modular compiler and toolchain technologies, runtime library
ii  libllvm9:amd64                             1:9-2~ubuntu18.04.2                              amd64        Modular compiler and toolchain technologies, runtime library
ii  libxkbcommon-dev:amd64                     0.8.2-1~ubuntu18.04.1                            amd64        library interface to the XKB compiler - development files
ii  libxkbcommon0:amd64                        0.8.2-1~ubuntu18.04.1                            amd64        library interface to the XKB compiler - shared library
```

My code looks like this:

**Code: test_1**
```cpp
#include <iostream>
#include <memory>
#include <opencv2/opencv.hpp>
#include <pcl/io/pcd_io.h>
#include <pcl/point_cloud.h>
#include "o3d3xx_camera.h"
#include "o3d3xx_framegrabber.h"
#include "o3d3xx_image.h"

using namespace pcl;
using namespace pcl::io;
using namespace std;
using namespace cv;

#define BOLD    "\033[1m"
#define RESET   "\033[0m"
#define RED     "\033[31m"
#define GREEN   "\033[32m"
#define YELLOW  "\033[33m"
#define BLUE    "\033[34m"
#define MAGENTA "\033[35m"
#define CYAN    "\033[36m"
#define WHITE   "\033[37m"

string PCDFilenameBase;

int getSavedPCD_(int id, pcl::PointCloud<pcl::PointXYZI>::Ptr& cloud_xyzi)
{

    if (PCDFilenameBase.empty())
    {
        cout<<RED<<"PCDFilenameBase not set !!"<<endl;
        exit(-1);
    }
    std::string filename;

    filename = PCDFilenameBase + std::to_string(id) + ".pcd";

    if (pcl::io::loadPCDFile (filename, *cloud_xyzi) == -1) //load the file
    {
        cout<<RED<<"Error!! Couldn't read file"<<filename<<RESET<<endl;
        exit (-1);
    }
    cout<<GREEN<<"Loaded image "<<filename<<RESET<<"\n";
    return 1;
}
[... 136 lines cut ...]
    o3d3xx::ImageBuffer::Ptr img = std::make_shared<o3d3xx::ImageBuffer>();
    o3d3xx::FrameGrabber::Ptr fg = std::make_shared<o3d3xx::FrameGrabber>(cam, o3d3xx::IMG_AMP|o3d3xx::IMG_RDIS|o3d3xx::IMG_CART);
    o3d3xx::Logging::Init();
    fg->SWTrigger();
    if (! fg->WaitForFrame(img.get(), 1000))
    {
        std::cerr << "Timeout waiting for camera!" << std::endl;
        exit (-1);
    }
    cout<<"Captured pcd"<<endl;
}

int main(int argc, char** argv)
{
    testCapture("192.168.0.25");
    return 1;
}
```
Output of the above code (Code: test_2), no runtime error occurred.
```
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0725 17:30:03.503839 23650 frame_grabber.cpp:72] Camera connection info: ip=192.168.0.25, port=50010
Captured pcd

```

theseankelly, 2020-07-27:
Hi @robinbansal08 thanks for the detailed bug report.

I'll take a look at reproducing this. Are your dependencies (PCL, OpenCV, etc) all the versions available in the Ubuntu 18.04 apt repositories, or have you built anything custom for your application?

robinbansal08, 2020-07-28:
@theseankelly Thanks a lot for your reply.

No, some of the dependencies I took from GitHub and build them.

PCL            1.9.1
OpenCV     3.3.1
libo3d3xx    0.7.7
VTK            8.2.0

Rest of the dependencies, I have installed using apt-get.

theseankelly, 2020-07-28:
Hey @robinbansal08 based on your valgrind output it feels like this is something in the PCL/Eigen layer, not libo3d3xx.

In fact, this brought up some distant memories of encountering a similar issue back when I was porting some unrelated project from 16.04 to 18.04. I did some digging and sure enough I commented on the PCL bug:

https://github.com/PointCloudLibrary/pcl/issues/2364

This looks to be very similar to what you're observing. And also looks like the maintainers published the necessary compile flags as a cmake target: https://github.com/PointCloudLibrary/pcl/pull/2100

This is really stretching my memory but based on my comments on that PCL bug, you might try removing `PCL_DEFINITIONS` from the image module here:
https://github.com/ifm/libo3d3xx/blob/c4c89730b47f765f9d303a2a4beeb8b922addcd3/modules/image/CMakeLists.txt#L148

And re-build.

Let me know whether that helps and I can push a patch.

theseankelly, 2020-07-28:
And possibly add in 
```
target_compile_options(${PROJECT_NAME} PUBLIC ${PCL_COMPILE_OPTIONS})
```
To the shared/static targets:
https://github.com/ifm/libo3d3xx/blob/master/modules/image/src/libo3d3xx_image/CMakeLists.txt

But I didn't have to do this back when I resolved the bug in my other project..

robinbansal08, 2020-07-29:
Based on your suggestions I made the changes but I have encountered cmake error while building image module.

**CMAKE error**
`target_compile_options(${PROJECT_NAME} PUBLIC ${PCL_COMPILE_OPTIONS})`
After adding this line in libo3d3xx/modules/image/src/libo3d3xx_image/CMakeLists.txt, I am getting cmake error as mentioned below

`
-- UBUNTU_VERSION: 18.04.4
-- UBUNTU_MAJOR: 18
-- UBUNTU_MINOR: 04
-- UBUNTU_PATCH: 4
-- Eigen found (include: /usr/include/eigen3, version: 3.3.4)
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   date_time
--   iostreams
--   serialization
--   chrono
--   atomic
--   regex
-- looking for PCL_COMMON
-- OpenCV_DIR: /usr/local/share/OpenCV
CMake Error at src/libo3d3xx_image/CMakeLists.txt:16 (target_compile_options):
  Cannot specify compile options for target "O3D3XX_IMAGE" which is not built
  by this project.

-- Skipping o3d3xx-viewer
CMake Warning at CMakeLists.txt:170 (message):
  Skipping unit tests

-- DEB_FILE: libo3d3xx-image_0.7.7_ubuntu-18.04.4_amd64.deb
-- Configuring incomplete, errors occurred!
See also "/home/cloudlinkedge/Libraries/libo3d3xx-0.7.7/modules/image/build/CMakeFiles/CMakeOutput.log".
See also "/home/cloudlinkedge/Libraries/libo3d3xx-0.7.7/modules/image/build/CMakeFiles/CMakeError.log".
`

If I skip the above process of adding compiler options line `target_compile_options(${PROJECT_NAME} PUBLIC ${PCL_COMPILE_OPTIONS})`  in libo3d3xx/modules/image/src/libo3d3xx_image/CMakeLists.txt, and just  remove PCL_DEFINITIONS from the image module libo3d3xx/modules/image/CMakeLists.txt, image module successfully builds but **still getting double free or corruption (out) error**

robinbansal08, 2021-03-01:
Any feedback for this issue please.

robinbansal08, 2021-04-16:
I tried to debug the following code with  gdb debugger. 

**Code**
```cpp
#include <iostream>
#include <memory>
#include <opencv2/opencv.hpp>
#include <pcl/io/pcd_io.h>
#include <pcl/point_cloud.h>
#include "o3d3xx_camera.h"
#include "o3d3xx_framegrabber.h"
#include "o3d3xx_image.h"

using namespace pcl;
using namespace pcl::io;
using namespace std;
using namespace cv;

#define BOLD    "\033[1m"
#define RESET   "\033[0m"
#define RED     "\033[31m"
#define GREEN   "\033[32m"
#define YELLOW  "\033[33m"
#define BLUE    "\033[34m"
#define MAGENTA "\033[35m"
#define CYAN    "\033[36m"
#define WHITE   "\033[37m"

string PCDFilenameBase;

int getSavedPCD_(int id, pcl::PointCloud<pcl::PointXYZI>::Ptr& cloud_xyzi)
{

    if (PCDFilenameBase.empty())
    {
        cout<<RED<<"PCDFilenameBase not set !!"<<endl;
        exit(-1);
    }
    std::string filename;

    filename = PCDFilenameBase + std::to_string(id) + ".pcd";

    if (pcl::io::loadPCDFile (filename, *cloud_xyzi) == -1) //load the file
    {
        cout<<RED<<"Error!! Couldn't read file"<<filename<<RESET<<endl;
        exit (-1);
    }
    cout<<GREEN<<"Loaded image "<<filename<<RESET<<"\n";
    return 1;
}

void testCapture(string camIP)
{

    o3d3xx::Camera::Ptr cam = std::make_shared<o3d3xx::Camera>(camIP);
    o3d3xx::ImageBuffer::Ptr img = std::make_shared<o3d3xx::ImageBuffer>();
    o3d3xx::FrameGrabber::Ptr fg = std::make_shared<o3d3xx::FrameGrabber>(cam, o3d3xx::IMG_AMP|o3d3xx::IMG_RDIS|o3d3xx::IMG_CART);
    o3d3xx::Logging::Init();
    fg->SWTrigger();
    if (! fg->WaitForFrame(img.get(), 1000))
    {
        std::cerr << "Timeout waiting for camera!" << std::endl;
        exit (-1);
    }
    cout<<"Captured pcd"<<endl;
}

int main(int argc, char** argv)
{
    testCapture("192.168.0.25");
[... 60 lines cut ...]
#5  __GI___libc_free (mem=0x7fffe71c6020) at malloc.c:3124
#6  0x00007ffff770eb1f in o3d3xx::ImageBuffer::~ImageBuffer() () from /usr/lib/libo3d3xx_image.so.0.7.7
#7  0x0000555555588f7b in __gnu_cxx::new_allocator<o3d3xx::ImageBuffer>::destroy<o3d3xx::ImageBuffer> (this=0x555555824a90, __p=0x555555824a90) at /usr/include/c++/5/ext/new_allocator.h:124
#8  0x0000555555588d6f in std::allocator_traits<std::allocator<o3d3xx::ImageBuffer> >::destroy<o3d3xx::ImageBuffer> (__a=..., __p=0x555555824a90) at /usr/include/c++/5/bits/alloc_traits.h:542
#9  0x0000555555587e3f in std::_Sp_counted_ptr_inplace<o3d3xx::ImageBuffer, std::allocator<o3d3xx::ImageBuffer>, (__gnu_cxx::_Lock_policy)2>::_M_dispose (this=0x555555824a80)
    at /usr/include/c++/5/bits/shared_ptr_base.h:531
#10 0x000055555557fe6e in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=0x555555824a80) at /usr/include/c++/5/bits/shared_ptr_base.h:150
#11 0x000055555557ee61 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count (this=0x7fffffffdc88, __in_chrg=<optimized out>) at /usr/include/c++/5/bits/shared_ptr_base.h:659
#12 0x000055555557e980 in std::__shared_ptr<o3d3xx::ImageBuffer, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr (this=0x7fffffffdc80, __in_chrg=<optimized out>)
    at /usr/include/c++/5/bits/shared_ptr_base.h:925
#13 0x000055555557e99c in std::shared_ptr<o3d3xx::ImageBuffer>::~shared_ptr (this=0x7fffffffdc80, __in_chrg=<optimized out>) at /usr/include/c++/5/bits/shared_ptr.h:93
#14 0x000055555557c055 in testCapture (camIP="192.168.0.25") at /home/cloudlinkedge/xpsSea/speyerIPC_1/testLibo3d/testLibo3d.cpp:52
#15 0x000055555557c109 in main (argc=1, argv=0x7fffffffde18) at /home/cloudlinkedge/xpsSea/speyerIPC_1/testLibo3d/testLibo3d.cpp:66
(gdb) info frame
Stack level 0, frame at 0x7fffffffd840:
 rip = 0x7ffff567bf47 in __GI_raise (../sysdeps/unix/sysv/linux/raise.c:51); saved rip = 0x7ffff567d8b1
 called by frame at 0x7fffffffd970
 source language c.
 Arglist at 0x7fffffffd718, args: sig=sig@entry=6
 Locals at 0x7fffffffd718, Previous frame's sp is 0x7fffffffd840
 Saved registers:
  rip at 0x7fffffffd838
(gdb) 

```

graugans, 2021-04-19:
@robinbansal08, thanks for the feedback. I have assigned this issue to one of our developers.

graugans, 2021-04-19:
@robinbansal08 due to the fact it is not easy to reproduce you set-up, could you please check this [SO Answer](https://stackoverflow.com/a/2903152)? From the comments `export MALLOC_CHECK_=3` is recommended. Maybe `export MALLOC_CHECK_=2` is a work arround in case it does not crash your application.

Can you please also check if you may have installed both PCL from the Ubuntu apt-repositories and by compiling it on your own? 

Are you using CMake or any other tools to compile your application?

graugans, 2021-04-19:
> Based on your suggestions I made the changes but I have encountered cmake error while building image module.
>
> **CMAKE error**
> `target_compile_options(${PROJECT_NAME} PUBLIC ${PCL_COMPILE_OPTIONS})` [...]

The error is caused due to to the fact you cannot use `${PROJECT_NAME} ` IMHO. Better add the definitions [here](https://github.com/ifm/libo3d3xx/blob/master/modules/image/src/libo3d3xx_image/CMakeLists.txt#L33) for the shared library output and [here](https://github.com/ifm/libo3d3xx/blob/master/modules/image/src/libo3d3xx_image/CMakeLists.txt#L63) for the static one.

```cmake
target_compile_definitions(o3d3xx_image_shared PUBLIC ${PCL_DEFINITIONS})
target_compile_options(o3d3xx_image_shared PUBLIC ${PCL_COMPILE_OPTIONS})
# ...
target_compile_definitions(o3d3xx_image_static PUBLIC ${PCL_DEFINITIONS})
target_compile_options(o3d3xx_image_static PUBLIC ${PCL_COMPILE_OPTIONS})
```

Since the mentioned bug report also suggests to remove the [Line#L148 in modules/image/CMakeLists.txt](https://github.com/ifm/libo3d3xx/blob/master/modules/image/CMakeLists.txt#L148) I would give this a try as-well. Since this is all depending on the overall project please test this with example code first.

graugans, 2021-04-23:
> @theseankelly Thanks a lot for your reply.
> 
> No, some of the dependencies I took from GitHub and build them.
> 
> PCL 1.9.1
> OpenCV 3.3.1
> libo3d3xx 0.7.7
> VTK 8.2.0
> 
> Rest of the dependencies, I have installed using apt-get.

Hey @robinbansal08 can you please confirm the combination of VTK 8.2.0 and OpenCV 3.3.1 cause I get the following compile error when trying to reproduce your set-up:

```
#16 37.37 ../modules/viz/src/clouds.cpp: In constructor 'cv::viz::WCloud::WCloud(cv::InputArray, cv::InputArray, cv::InputArray)':                                                            
#16 37.37 ../modules/viz/src/clouds.cpp:80:13: error: 'class vtkPolyDataMapper' has no member named 'ImmediateModeRenderingOff'                                                               
#16 37.37      mapper->ImmediateModeRenderingOff();                                                                                                                                           
#16 37.37              ^~~~~~~~~~~~~~~~~~~~~~~~~                                                                                                                                              
#16 37.37 ../modules/viz/src/clouds.cpp: In constructor 'cv::viz::WPaintedCloud::WPaintedCloud(cv::InputArray)':                                                                              #16 37.37 ../modules/viz/src/clouds.cpp:120:13: error: 'class vtkPolyDataMapper' has no member named 'ImmediateModeRenderingOff'                                                              #16 37.37      mapper->ImmediateModeRenderingOff();                                                                                                                                           #16 37.37              ^~~~~~~~~~~~~~~~~~~~~~~~~                                                                                                                                              #16 37.37 ../modules/viz/src/clouds.cpp: In constructor 'cv::viz::WPaintedCloud::WPaintedCloud(cv::InputArray, const Point3d&, const Point3d&)':                                              
#16 37.37 ../modules/viz/src/clouds.cpp:146:13: error: 'class vtkPolyDataMapper' has no member named 'ImmediateModeRenderingOff'                                                              
#16 37.37      mapper->ImmediateModeRenderingOff();                                                                                                                                           
#16 37.37              ^~~~~~~~~~~~~~~~~~~~~~~~~
#16 37.37 ../modules/viz/src/clouds.cpp: In constructor 'cv::viz::WPaintedCloud::WPaintedCloud(cv::InputArray, const Point3d&, const Point3d&, const cv::viz::Color&, cv::viz::Color)':
#16 37.37 ../modules/viz/src/clouds.cpp:185:13: error: 'class vtkPolyDataMapper' has no member named 'ImmediateModeRenderingOff'
#16 37.37      mapper->ImmediateModeRenderingOff();
#16 37.37              ^~~~~~~~~~~~~~~~~~~~~~~~~
#16 37.37 ../modules/viz/src/clouds.cpp: In constructor 'cv::viz::WCloudCollection::WCloudCollection()':
#16 37.37 ../modules/viz/src/clouds.cpp:214:13: error: 'class vtkPolyDataMapper' has no member named 'ImmediateModeRenderingOff'
#16 37.37      mapper->ImmediateModeRenderingOff();
#16 37.37              ^~~~~~~~~~~~~~~~~~~~~~~~~
#16 37.37 ../modules/viz/src/clouds.cpp: In constructor 'cv::viz::WMesh::WMesh(const cv::viz::Mesh&)':
```

robinbansal08, 2021-04-23:
> Hey @robinbansal08 can you please confirm the combination of VTK 8.2.0 and OpenCV 3.3.1 cause I get the following compile error when trying to reproduce your set-up:
>
> ```
> #16 37.37 [...]

Yes I have same versions of opencv and vtk. Are you getting this error while PCL installation ?

robinbansal08, 2021-04-23:
> The error is caused due to to the fact you cannot use `${PROJECT_NAME} ` IMHO. Better add the definitions [...]

I have tried these modifications, I was able to build the image module successfully. But still getting the same double free or corruption fault for my code.

graugans, 2021-04-23:
Thanks for the test, the compilation error I do receive while building OpenCV, as a start I disabled the VTK Option for OpenCV

graugans, 2021-04-26:
Some update on the current status, I have set-up an Ubuntu 18.04 based Docker image with the described dependencies build from source:

```docker
ARG UBUNTU_VERSION=18.04

FROM ubuntu:${UBUNTU_VERSION} AS buildstage

# Set none interactive mode when running in Docker
ENV DEBIAN_FRONTEND noninteractive

# change this if zour camera does have a different ip setting
ENV O3D3XX_IP 192.168.0.69

# Version Tags

ENV PCL_VERSION pcl-1.9.1
ENV OPENCV_VERSION 3.3.1
ENV VTK_VERSION v8.2.0
ENV LIBO3D3XX_VERSION v0.7.7

ENV BUILD_TYPE Debug

RUN apt-get update && apt-get install  --no-install-recommends -y \
    git \
    libcurl4-openssl-dev \
    libboost-all-dev \
    libgoogle-glog-dev \
    libxmlrpc-c++8-dev \
    libeigen3-dev \
    libflann-dev \
    libgtest-dev \
    libavcodec-dev \
    libavformat-dev \
    libavutil-dev \
    libdouble-conversion-dev \
    libexpat1-dev \
    libfontconfig-dev \
    libfreetype6-dev \
    libgdal-dev \
    libglew-dev \
    libhdf5-dev \
    libjpeg-dev \
    libjsoncpp-dev \
    liblz4-dev \
    liblzma-dev \
    libnetcdf-dev \
    libnetcdf-cxx-legacy-dev \
    libogg-dev \
    libpng-dev \
    libpython3-dev \
    libqt5opengl5-dev \
    libqt5x11extras5-dev \
    libsqlite3-dev \
    libswscale-dev \
    libtheora-dev \
    libtiff-dev \
    libxml2-dev \
    libxt-dev \
    qtbase5-dev \
    qttools5-dev \
    zlib1g-dev \
    coreutils \
    ca-certificates \
    iputils-ping \
    build-essential \
    ninja-build \
    cmake \
    sudo \
    vim
    
[... 153 lines cut ...]
RUN sudo rm -rf VTK pcl opencv
```
This pulls [a copy of libo3d3xx](https://github.com/graugans/libo3d3xx/tree/bugfix/142) with some patches needed for Ubuntu 18.04 plus your example from above.

I was not yet able to reproduce the double free corruption but will try a little bit more. This is how I build [the example](https://github.com/graugans/libo3d3xx/blob/bugfix/142/modules/examples/CMakeLists.txt#L167)
```cmake
add_executable(issue-142 issue-142.cpp)
target_link_libraries(issue-142
                      ${O3D3XX_CAMERA_LIBRARIES}
                      ${O3D3XX_FRAMEGRABBER_LIBRARIES}
                      ${O3D3XX_IMAGE_LIBRARIES}
                      ${O3D3XX_PCICCLIENT_LIBRARIES}
                      ${OpenCV_LIBRARIES}
                      ${LIB_boost_system}
                      ${CMAKE_THREAD_LIBS_INIT}
                      )
# Deal with the PCL library
target_compile_definitions(issue-142 PUBLIC ${PCL_DEFINITIONS})
target_compile_options(issue-142 PUBLIC ${PCL_COMPILE_OPTIONS})
target_link_libraries(issue-142 ${PCL_LIBRARIES})
target_compile_definitions(issue-142 PUBLIC ${PCL_DEFINITIONS})
target_compile_options(issue-142 PUBLIC ${PCL_COMPILE_OPTIONS})
```

@robinbansal08 please let me know if you have compiled or configured your system in a different way.

robinbansal08, 2021-04-27:
> Some update on the current status, I have set-up an Ubuntu 18.04 based Docker image with the described dependencies build from source:
> 
> ```
> ARG UBUNTU_VERSION=18.04
> 
> FROM ubuntu:${UBUNTU_VERSION} AS buildstage
> 
> # Set none interactive mode when running in Docker
> ENV DEBIAN_FRONTEND noninteractive
> 
> # change this if zour camera does have a different ip setting
> ENV O3D3XX_IP 192.168.0.69
> 
> # Version Tags
> 
> ENV PCL_VERSION pcl-1.9.1
> ENV OPENCV_VERSION 3.3.1
> ENV VTK_VERSION v8.2.0
> ENV LIBO3D3XX_VERSION v0.7.7
> 
> ENV BUILD_TYPE Debug
> 
> RUN apt-get update && apt-get install  --no-install-recommends -y \
>     git \
>     libcurl4-openssl-dev \
>     libboost-all-dev \
>     libgoogle-glog-dev \
>     libxmlrpc-c++8-dev \
>     libeigen3-dev \
>     libflann-dev \
>     libgtest-dev \
>     libavcodec-dev \
>     libavformat-dev \
>     libavutil-dev \
>     libdouble-conversion-dev \
>     libexpat1-dev \
>     libfontconfig-dev \
>     libfreetype6-dev \
>     libgdal-dev \
>     libglew-dev \
>     libhdf5-dev \
>     libjpeg-dev \
>     libjsoncpp-dev \
>     liblz4-dev \
>     liblzma-dev \
>     libnetcdf-dev \
>     libnetcdf-cxx-legacy-dev \
>     libogg-dev \
>     libpng-dev \
>     libpython3-dev \
>     libqt5opengl5-dev \
>     libqt5x11extras5-dev \
>     libsqlite3-dev \
>     libswscale-dev \
>     libtheora-dev \
>     libtiff-dev \
>     libxml2-dev \
>     libxt-dev \
>     qtbase5-dev \
>     qttools5-dev \
>     zlib1g-dev \
>     coreutils \
>     ca-certificates \
>     iputils-ping \
>     build-essential \
>     ninja-build \
>     cmake \
>     sudo \
>     vim
>     
[... 166 lines cut ...]
>                       ${O3D3XX_PCICCLIENT_LIBRARIES}
>                       ${OpenCV_LIBRARIES}
>                       ${LIB_boost_system}
>                       ${CMAKE_THREAD_LIBS_INIT}
>                       )
> # Deal with the PCL library
> target_compile_definitions(issue-142 PUBLIC ${PCL_DEFINITIONS})
> target_compile_options(issue-142 PUBLIC ${PCL_COMPILE_OPTIONS})
> target_link_libraries(issue-142 ${PCL_LIBRARIES})
> target_compile_definitions(issue-142 PUBLIC ${PCL_DEFINITIONS})
> target_compile_options(issue-142 PUBLIC ${PCL_COMPILE_OPTIONS})
> ```
> 
> @robinbansal08 please let me know if you have compiled or configured your system in a different way.
> 
> **EDIT:**
> 
> ```
> o3d3xx@67cc76c76387:~/build/examples$ O3D3XX_IP=192.168.0.71 ./issue-142 
> WARNING: Logging before InitGoogleLogging() is written to STDERR
> I0426 18:17:52.290215    12 frame_grabber.cpp:72] Camera connection info: ip=192.168.0.71, port=50010
> Captured pcd
> ```

@graugans  Thanks a lot for your efforts. I have also compiled the same way but not configuring the cmake options of opencv, pcl and vtk like you did, which might have made the difference. I will try to install the libraries in a new system with similar cmake options.

graugans, 2021-04-27:
That is very likely, most of the flags I took from various tutorials on the internet for compiling the libraries. Maybe a few additional libraries and settings are needed depending on you use-case.

laiskumus, 2021-04-28:
@robinbansal08 @graugans I had a similar issue and it was resolved by building all libraries consistently with ”-mavx”, as explained here:  https://github.com/PointCloudLibrary/pcl/issues/2147#issuecomment-399950762

robinbansal08, 2021-05-20:
This issue is resolved. I did libraries installation as you did, I didn't get any error in frame-grabber process. Thanks a lot for your support.

closed 2021-05-20 by robinbansal08