ifm/libo3d3xx #144: Automatic user application startup during sensor power-up

robinbansal08, 2021-04-29:
I am trying to start my application during sensor boot process, I have referred to the already mentioned guidelines [https://github.com/ifm/libo3d3xx/issues/98](url). But still I am unable to add my application to boot process.

I have written the init script _startupTest.sh_ and kept in the location ` ` `/opt/oem/etc/init.d/` ` ` and my application binary file _cmdControl_Cam_ in the location ` ` `/opt/oem/usr/bin` ` `. After a reboot still my application cmdControl_Cam didn't start. I have noticed that startupTest.sh can be run successfully by using command  ` ` `bash  startupTest.sh start` ` `. Can you please provide any suggestions @graugans.
```
### BEGIN INIT INFO
# Provides: startupTest
# Required-Start: 
# Required-Stop: 
# Default-Start:  2 3 4 5
# Default-Stop: 0 1 6
# Short-Description: start cmd server
# Description: start cmd server
### END INIT INFO

case "$1" in
  start)
    cd ../../usr/bin/
    ./cmdControl_Cam &
    ;;
  stop)
    ;;
  *)
    exit 1
    ;;
esac

exit 0
```

graugans, 2021-04-29:
Most probably it is a difference between the [login-shell (ssh) and no-login shell](https://relentlesscoding.com/posts/whats-the-difference-between-a-login-and-a-nonlogin-shell/). Maybe you need to specify more environment variables. One thing you might not see your application running is the fact it crashed due to some libs or other stuff is missing.

To veryfiy that the script runs one could simply touch a file

```
touch /tmp/cmdControl_Cam.seen
```

graugans, 2021-04-29:
May I recommend to change your script like this:

```bash
### BEGIN INIT INFO
# Provides: startupTest
# Required-Start: 
# Required-Stop: 
# Default-Start:  2 3 4 5
# Default-Stop: 0 1 6
# Short-Description: start cmd server
# Description: start cmd server
### END INIT INFO
PIDFILE=/opt/oem/var/run/$NAME.pid
DAEMON=/opt/oem/usr/bin/cmdControl_Cam
DAEMON_OPTS=""

export O3D3XX_IP=127.0.0.1

case "$1" in
  start)
    mkdir -p /opt/oem/var/run
    touch /tmp/cmdControl_Cam.seen
    start-stop-daemon --start --quiet --pidfile $PIDFILE --exec $DAEMON -- $DAEMON_OPTS
    ;;
  stop)
   start-stop-daemon --stop --quiet --oknodo --pidfile $PIDFILE
    ;;
  *)
    exit 1
    ;;
esac

exit 0

robinbansal08, 2021-04-29:
> May I recommend to change your script like this:
> 
> ```shell
> ### BEGIN INIT INFO
> # Provides: startupTest
> # Required-Start: 
> # Required-Stop: 
> # Default-Start:  2 3 4 5
> # Default-Stop: 0 1 6
> # Short-Description: start cmd server
> # Description: start cmd server
> ### END INIT INFO
> PIDFILE=/opt/oem/var/run/$NAME.pid
> DAEMON=/opt/oem/usr/bin/cmdControl_Cam
> DAEMON_OPTS=""
> 
> export O3D3XX_IP=127.0.0.1
> 
> case "$1" in
>   start)
>     mkdir -p /opt/oem/var/run
>     touch /tmp/cmdControl_Cam.seen
>     start-stop-daemon --start --quiet --pidfile $PIDFILE --exec $DAEMON -- $DAEMON_OPTS
>     ;;
>   stop)
>    start-stop-daemon --stop --quiet --oknodo --pidfile $PIDFILE
>     ;;
>   *)
>     exit 1
>     ;;
> esac
> 
> exit 0
> ```
> 
> By using `./cmdControl_Cam &` you fork into the background, but the calling shell is terminated which also typically terminates its childs.

I have tried the suggested init script, but still it doesn't start my application during bootup process.

graugans, 2021-04-29:
Are you able to start the script manually, I just wrote it from top of my head, is the file `/tmp/cmdControl_Cam.seen` generated? In case your application is writing anything to stdout and stderr it may help to redirect both to a file to see what is going on....

robinbansal08, 2021-04-29:
> Are you able to start the script manually, I just wrote it from top of my head, is the file `/tmp/cmdControl_Cam.seen` generated? In case your application is writing anything to stdout and stderr it may help to redirect both to a file to see what is going on....

I am getting error when I try to the run the suggested init script manually.
```
oem@O3D303-41-39-20:/opt/oem/etc/init.d$ bash startupTest.sh start
startupTest.sh: line 20: start-stop-daemon: command not found
```

graugans, 2021-04-29:
You have to add `/sbin` to the PATH
```
PATH=/sbin:$PATH
```
or run 
```
/sbin/start-stop-daemon
```

robinbansal08, 2021-04-29:
> You have to add `/sbin` to the PATH
> 
> ```
> PATH=/sbin:$PATH
> ```
> 
> or run
> 
> ```
> /sbin/start-stop-daemon
> ```

It worked with your suggested changes. Thanks a lot for your support.

closed 2021-04-29 by robinbansal08